rbonghi/jetson_stats #427: Installation of jetson_stats_4.2.2 on nvidia-jetson-nano-4GB is stuck at: INFO:jtop.service: - ENABLE jtop.service

flomaurer, 2023-06-29:
## Describe the bug

When trying to install jetson_stats on our Nvidia nano 4GB Board, which runs duckietown (https://docs.duckietown.com/daffy/opmanual-duckiebot/setup/setup_sd_card/index.html) the installation gets stuck at 'INFO:jtop.service: - ENABLE jtop.service'

## To Reproduce

Steps to reproduce the behavior:

1. python3 --version
Python 3.7.2
2. python3 -m pip --version
pip 23.1.2 from /home/duckie/archiconda3/lib/python3.7/site-packages/pip (python 3.7)
3. wget https://github.com/rbonghi/jetson_stats/archive/refs/tags/4.2.2.zip
4. See error: sudo -H python3 -m pip install ./4.2.2.zip -v
5. journal also just mentiones, that starting jtop.service failed; also a manual start 'sudo systemctl restart jtop.service' doesn't work; even after a reboot the bahaviour doesn't change

## Screenshots

/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography. The next release of cryptography will remove support for Python 3.6.
  from cryptography.utils import int_from_bytes
Using pip 21.3.1 from /usr/local/lib/python3.6/dist-packages/pip (python 3.6)
Processing ./4.2.2.zip
  Running command python setup.py egg_info
  running egg_info
  creating /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info
  writing /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/SOURCES.txt'
  /usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  INFO:root:Generating grammar tables from /usr/lib/python3.6/lib2to3/Grammar.txt
  INFO:root:Generating grammar tables from /usr/lib/python3.6/lib2to3/PatternGrammar.txt
  reading manifest file '/tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files found matching 'tox.ini'
  warning: no previously-included files found matching 'Dockerfile'
  warning: no previously-included files found matching '.dockerignore'
  no previously-included directories found matching 'tests'
  no previously-included directories found matching 'docs'
  no previously-included directories found matching 'examples'
  writing manifest file '/tmp/pip-pip-egg-info-4swwgk1n/jetson_stats.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Requirement already satisfied: distro in /usr/local/lib/python3.6/dist-packages (from jetson-stats==4.2.2) (1.8.0)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.6/dist-packages (from jetson-stats==4.2.2) (0.4.2)
Building wheels for collected packages: jetson-stats
  Running command /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-oeefw1mh/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-oeefw1mh/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-v8cgdfhw
  /usr/lib/python3.6/distutils/dist.py:261: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  running bdist_wheel
  running build
  running build_py
  INFO:root:Generating grammar tables from /usr/lib/python3.6/lib2to3/Grammar.txt
  INFO:root:Generating grammar tables from /usr/lib/python3.6/lib2to3/PatternGrammar.txt
  creating build
  creating build/lib
  creating build/lib/jtop
  copying jtop/__main__.py -> build/lib/jtop
  copying jtop/jetson_release.py -> build/lib/jtop
  copying jtop/__init__.py -> build/lib/jtop
  copying jtop/jetson_swap.py -> build/lib/jtop
  copying jtop/service.py -> build/lib/jtop
  copying jtop/github.py -> build/lib/jtop
  copying jtop/jetson_config.py -> build/lib/jtop
  copying jtop/terminal_colors.py -> build/lib/jtop
  copying jtop/jtop.py -> build/lib/jtop
  creating build/lib/jtop/core
  copying jtop/core/jetson_variables.py -> build/lib/jtop/core
[... 139 lines cut ...]

^C  Building wheel for jetson-stats (setup.py) ... canceled

## Expected behavior

the installation to finish to use jtop

## Additional context

when flashing the SD-Card with the NVIDIA toolchain I can install jtop on the very same board
but with the duckietown image it doesn't work

### Board

* jetson-stats version: 4.2.2
* Module: NVIDIA Jetson Nano 4GB (NVIDIA Jetson Nano Developer Kit)
* L4T: # R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020
* Ubuntu 18.04.6 LTS (GNU/Linux 4.9.140-tegra aarch64)

### Log from jtop.service
journalctl -u jtop.service -n 100 --no-pager:
Failed to start jtop.service.

### Log from jetson-stats installation
see above

flomaurer, 2023-06-29:
Thanks for this hint. I'm able to install v4.1.0 and also v4.1.5 (only tested these two out of the v4.1.X series).
It seems like the problem comes with the v4.2.X series (v4.2.0 and v4.2.2 are stuck as described above).

So it would be great, if in the future also the latest releases which are selected when doing `sudo -H python3 -m pip install jetson_stats -v` get installed without problems.

LudeeD, 2024-01-23:
I'm also experience this issue while installing in a Jetson Orin NX 8GB, the pip instalation completes, but at the end no jtop group, no jtop service, jtop does not work

ProToTime3560, 2024-01-23:
i think this version is error version. to install. there is no jtop group, no jtop service. 
in my case, i found jetson_config  jetson_release  jetson_swap  jtop file in /usr/local/bin. when i try sudo pip3 install jetson-stats
if you install this version. you can't jtop in terminal when you reboot or sudo systemctl restart.
i found one case to run sudo jtop. follow
1.sudo pip3 install jetson-stats
2.sudo find / -name jtop.service ( you can find where is jtop.py is)
3.sudo mv /usr/local/jetson_stats/jtop.service /etc/systemd/system/ (for input jtop.service)
  or you can try sudo ln -s /usr/local/jetson_stats/jtop.service /etc/systemd/system/jtop.service
4.sudo systemctl daemon-reload
5.sudo systemctl start jtop.service
6.sudo systemctl enable jtop.service
7.sudo systemctl status jtop.service (then you can see service status. and now you need to reboot.
8.sudo jtop (after reboot).
9.sudo jtop --health (you can try fix. but i catch there is not file for fix).
i hope you can run jtop. but good way is install downgrade version or wait for next version for fix bug.

rbonghi, 2024-01-24:
thank you to write this message, I think the error is the same on this issue #488 and #470

Can you please share me the output from 

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

```
journalctl -u jtop.service -n 100 --no-pager
```

if the output from the latest command is empty can you please me share also the output from this command:
```
sudo JTOP_SERVICE=True jtop --force
```

danielzhangau, 2024-01-25:
```
hi@localhost:~$ journalctl -u jtop.service -n 100 --no-pager
-- Logs begin at Thu 2024-01-25 04:03:05 UTC, end at Thu 2024-01-25 04:52:43 UTC. --
-- No entries --
hi@localhost:~$ sudo JTOP_SERVICE=True jtop --force
[INFO] jtop.core.config - Build service folder in /usr/local/jtop
[INFO] jtop.service - jetson_stats 4.2.5 - server loaded
[INFO] jtop.service - Running on Python: 3.6.9
[INFO] jtop.core.hardware - Hardware detected aarch64
[INFO] jtop.core.hardware - NVIDIA Jetson 699-level Part Number=699-13448-0002-401 G.0
[INFO] jtop.core.hardware - NVIDIA Jetson Module=NVIDIA Jetson Nano module (16Gb eMMC)
[INFO] jtop.core.hardware - NVIDIA Jetson detected L4T=32.5.1
[INFO] jtop.core.cpu - Found 4 CPU
[INFO] jtop.core.gpu - GPU "gpu" status in /sys/devices/57000000.gpu
[INFO] jtop.core.gpu - GPU "gpu" frq in /sys/devices/57000000.gpu/devfreq/57000000.gpu
[INFO] jtop.core.processes - Process service started
[INFO] jtop.core.memory - Found EMC!
[INFO] jtop.core.memory - Found IRAM!
[INFO] jtop.core.memory - Memory service started
[INFO] jtop.core.engine - Engines found: [APE NVDEC NVENC NVJPG SE VIC]
[INFO] jtop.core.temperature - Found thermal "PLL" in thermal_zone3
[INFO] jtop.core.temperature - Found thermal "CPU" in thermal_zone1
[WARNING] jtop.core.temperature - Skipped PMIC
[INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone2
[INFO] jtop.core.temperature - Found thermal "AO" in thermal_zone0
[INFO] jtop.core.temperature - Found thermal "thermal" in thermal_zone5
[INFO] jtop.core.power - Found I2C power monitor
[INFO] jtop.core.fan - Fan tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
[INFO] jtop.core.fan - RPM tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
[INFO] jtop.core.fan - Fan temp controller tegra_pwmfan found in /sys/class/hwmon/hwmon1/temp_control
[INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats==4.2.5', 'console_scripts', 'jtop')()
  File "/usr/local/lib/python3.6/dist-packages/jtop/__main__.py", line 104, in main
    server = JtopServer(force=args.force)
  File "/usr/local/lib/python3.6/dist-packages/jtop/service.py", line 302, in __init__
    self.nvpmodel = NVPModelService(self.jetson_clocks)
  File "/usr/local/lib/python3.6/dist-packages/jtop/core/nvpmodel.py", line 441, in __init__
    name=self._nvpmodel_now['name'],
TypeError: 'NoneType' object is not subscriptable
```
The TypeError can be fixed by configuring the Power Model (NVPModel):
`sudo nvpmodel -m 0
`
Try `sudo JTOP_SERVICE=True jtop --force` again:
```
vhq@localhost:~$ sudo JTOP_SERVICE=True jtop --force
[INFO] jtop.service - jetson_stats 4.2.5 - server loaded
[INFO] jtop.service - Running on Python: 3.6.9
[INFO] jtop.core.hardware - Hardware detected aarch64
[INFO] jtop.core.hardware - NVIDIA Jetson 699-level Part Number=699-13448-0002-401 G.0
[INFO] jtop.core.hardware - NVIDIA Jetson Module=NVIDIA Jetson Nano module (16Gb eMMC)
[INFO] jtop.core.hardware - NVIDIA Jetson detected L4T=32.5.1
[INFO] jtop.core.cpu - Found 4 CPU
[INFO] jtop.core.gpu - GPU "gpu" status in /sys/devices/57000000.gpu
[INFO] jtop.core.gpu - GPU "gpu" frq in /sys/devices/57000000.gpu/devfreq/57000000.gpu
[INFO] jtop.core.processes - Process service started
[INFO] jtop.core.memory - Found EMC!
[INFO] jtop.core.memory - Found IRAM!
[INFO] jtop.core.memory - Memory service started
[INFO] jtop.core.engine - Engines found: [APE NVDEC NVENC NVJPG SE VIC]
[INFO] jtop.core.temperature - Found thermal "PLL" in thermal_zone3
[INFO] jtop.core.temperature - Found thermal "CPU" in thermal_zone1
[WARNING] jtop.core.temperature - Skipped PMIC
[INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone2
[INFO] jtop.core.temperature - Found thermal "AO" in thermal_zone0
[INFO] jtop.core.temperature - Found thermal "thermal" in thermal_zone5
[INFO] jtop.core.power - Found I2C power monitor
[INFO] jtop.core.fan - Fan tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
[... 15 lines cut ...]
[INFO] jtop.core.hardware - NVIDIA Jetson detected L4T=32.5.1
[INFO] jtop.core.cpu - Found 4 CPU
[INFO] jtop.core.gpu - GPU "gpu" status in /sys/devices/57000000.gpu
[INFO] jtop.core.gpu - GPU "gpu" frq in /sys/devices/57000000.gpu/devfreq/57000000.gpu
[INFO] jtop.core.processes - Process service started
[INFO] jtop.core.memory - Found EMC!
[INFO] jtop.core.memory - Found IRAM!
[INFO] jtop.core.memory - Memory service started
[INFO] jtop.core.engine - Engines found: [APE NVDEC NVENC NVJPG SE VIC]
[INFO] jtop.core.temperature - Found thermal "PLL" in thermal_zone3
[INFO] jtop.core.temperature - Found thermal "CPU" in thermal_zone1
[WARNING] jtop.core.temperature - Skipped PMIC
[INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone2
[INFO] jtop.core.temperature - Found thermal "AO" in thermal_zone0
[INFO] jtop.core.temperature - Found thermal "thermal" in thermal_zone5
[INFO] jtop.core.power - Found I2C power monitor
[INFO] jtop.core.fan - Fan tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
[INFO] jtop.core.fan - RPM tegra_pwmfan(1) found in /sys/class/hwmon/hwmon1
[INFO] jtop.core.fan - Fan temp controller tegra_pwmfan found in /sys/class/hwmon/hwmon1/temp_control
[INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
[INFO] jtop.core.nvpmodel - nvpmodel running in [0]MAXN - Default: 0
[INFO] jtop.service - Initialization service
[INFO] jtop.core.jetson_clocks - Store jetson_clocks configuration in /usr/local/jtop/l4t_dfs.conf
[INFO] jtop.service - service ready
```

rbonghi, 2024-01-25:
Can you test this release candidate for me? Looks fixed @LudeeD @ProToTime3560 @danielzhangau 

```
sudo -H pip3 install -v jetson-stats==4.2.6rc1
```

if you confirm I'll make a new version stable
Otherwise can you copy and paste the output of this installation

Thank you very much

rbonghi, 2024-01-25:
Fixed with new release **4.2.6**

Update with:
```
sudo -H pip3 install -U jetson-stats
```

closed 2024-01-25 by rbonghi