rbonghi/jetson_stats #428: [Jetson AGX Orin 64 GB] jetson stats not starting - Error with GUI colors

laminair, 2023-07-03:
## Describe the bug

I just installed jetson_stats on a bunch of Jetson AGX Orin 64 GB devices. Tried installation via pip and from source (GIT repo).
Launched the systemd service without error but cannot access the GUI.

## To Reproduce

**Installation via pip** 
```
sudo -H pip3 install --no-cache-dir -U jetson-stats
```

**Installation from source**
```
git clone https://github.com/rbonghi/jetson_stats.git
sudo python3 setup.py build
sudo python3 setup.py install
```

Steps to reproduce the behavior:

Run `jtop` or `sudo jtop`. Both produces the same error

## Screenshots / Terminal Logs

Running `jtop`: 
```
ubuntu@orin-02:~/jetson_stats$ jtop

Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats==4.2.2', 'console_scripts', 'jtop')()
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 159, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 79, in __init__
    NColors(color_filter)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/lib/colors.py", line 43, in __init__
    curses.init_pair(NColors.RED, curses.COLOR_RED if not color_filter else curses.COLOR_BLUE, curses.COLOR_BLACK)
_curses.error: init_pair() returned ERR
```

Running `sudo jtop --health`: 
```
ubuntu@orin-02:~/jetson_stats$ sudo jtop --health
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 11, in <module>
    load_entry_point('jetson-stats==4.2.2', 'console_scripts', 'jtop')()
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 134, in main
    jtop_config()
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_config.py", line 227, in jtop_config
    curses.wrapper(JTOPCONFIG, JTOP_MENU)
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopguiconfig.py", line 39, in __init__
    curses.init_pair(1, curses.COLOR_RED, curses.COLOR_BLACK)
_curses.error: init_pair() returned ERR
```

## Expected behavior

When running `jtop`, the jetson_stats GUI becomes available.

## Additional context

From the logs, the jetson_stats service seems to be running just fine.
I already set the power profile to max power, as suggested by #406.  

### Board
[... 121 lines cut ...]
Jul 03 14:12:05 orin-02 jtop[3569]: [WARNING] jtop.core.power - Skipped "sum of shunt voltages" /sys/bus/i2c/devices/1-0040/hwmon/hwmon3/in7_label
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.power - Found I2C power monitor
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.power - Found name=1-00081 type=USB model=<EMPTY> in ucsi-source-psy-1-00081
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.power - Found name=1-00082 type=USB model=<EMPTY> in ucsi-source-psy-1-00082
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.power - Found SYSTEM power monitor
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.fan - Fan pwmfan(1) found in /sys/class/hwmon/hwmon2
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.fan - RPM pwm_tach found in /sys/class/hwmon/hwmon0
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.fan - Found nvfancontrol.service
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
Jul 03 14:12:05 orin-02 jtop[3569]: [INFO] jtop.core.nvpmodel - nvpmodel running in [0]MAXN - Default: 2
Jul 03 14:12:05 orin-02 jtop[3588]: [INFO] jtop.service - Initialization service
Jul 03 14:12:06 orin-02 jtop[3588]: [INFO] jtop.service - service ready
Jul 03 14:12:07 orin-02 jtop[3588]: [INFO] jtop.service - jtop timer thread started 1000ms
Jul 03 14:12:11 orin-02 jtop[3588]: [INFO] jtop.service - jtop timer thread close
Jul 03 14:17:17 orin-02 jtop[3588]: [INFO] jtop.service - jtop timer thread started 1000ms
Jul 03 14:17:21 orin-02 jtop[3588]: [INFO] jtop.service - jtop timer thread close
```

### Log from jetson-stats installation

```
Requirement already up-to-date: jetson-stats in /usr/local/lib/python3.8/dist-packages (4.2.2)
Requirement already satisfied, skipping upgrade: distro in /usr/lib/python3/dist-packages (from jetson-stats) (1.4.0)
Requirement already satisfied, skipping upgrade: smbus2 in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (0.4.2)
```

NguyenKhanh27, 2023-07-08:
I saw no one support on this so I did compare the system service changes from v4.1 -> 4.2 is that they add "After=multi-user.target", tbh I'm not familiar with system service but in my case, I rm that line in jtop system service and then it works for me. Maybe you can have a try

laminair, 2023-07-08:
@NguyenKhanh27 Thanks for the suggestion! That did the job but it seems odd that there is a conflict between multi-user.target (which is the user console/GUI being available) and the Jtop GUI (colors, to be specific).

Anyway, it works. Thanks.

closed 2023-07-08 by laminair