rbonghi/jetson_stats #435: jetson-stats not supported for [L4T 35.4.1]

BartoszKossowski, 2023-08-04:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 35.4.1

### Jetson-Stats

 - Version: 4.2.2
From terminal:
**[WARN] jetson-stats not supported for [L4T 35.4.1]**

dudo, 2023-08-10:
Any idea on when you'll have a chance to update this?

atinfinity, 2023-08-11:
I add the following value to [jtop/core/jetson_variables.py](https://github.com/rbonghi/jetson_stats/blob/4.2.2/jtop/core/jetson_variables.py#L43).

```
"35.4.1": "5.1.2",
```

As a result, I can show the information of L4T 35.4.1.

### jetson_release

[image]

### jtop

If you changed code, you need to restart `jtop` service.

[image]

rbonghi, 2023-08-14:
Fixed with the new release. Please update to **4.2.3**
```
sudo pip3 install -U jetson-stats
```

closed 2023-08-14 by rbonghi