rbonghi/jetson_stats #464: Add NVME Temperature Monitoring

jmachuca77, 2023-11-01:
If the Jetson has an NVME installed it would be nice to be able to see the temperature from the NVME. We are having an issue with a custom carrier board where the NVME is overheating. NVME-CLI seems to have a solution for reading the temp out.

I have not found a real alternative for reading out the temperature in realtime.

rbonghi, 2024-01-04:
Hi @jmachuca77 

It's a feature that I was also thinking of adding on jtop, but the documentation on NVME devices is not clear!
There are libraries available but not in Python, or the licensing doesn't permit me to use it.

I couldn't find any official documentation to manually read these values from the OS, so I couldn't create my personal driver.
If you know references let me know.

Raffaello

LinCanNerd, 2026-10-12:
On JetPack 7.2.1 (L4T 39.2.0, Jetson AGX Thor), jtop already shows NVMe temperatures. The L4T kernel is built with `CONFIG_NVME_HWMON=y`, so the drive shows up in `/sys/class/hwmon` with the name `nvme`. jtop reads hwmon temperature sensors, and since 7.2.0 it labels them with the device name: `nvme Composite`, `nvme Sensor 1`, `nvme Sensor 2`.

On older JetPack releases, it depends on whether the kernel enables `CONFIG_NVME_HWMON`. You can check with `cat /sys/class/hwmon/hwmon*/name`.

Related: #924 fixes a bug where NVMe sensors without a `crit` threshold were always drawn red.

If this covers your use case, I think this issue can be closed.