rbonghi/jetson_stats #466: Jtop service not installed correctly on Jetson Nano

halsimov, 2023-11-15:
## Describe the bug

jtop is installed but do act as if the jtop.service is not enabled

## To Reproduce

Steps to reproduce the behavior:

1. Install jtop using ` sudo pip3 install  -U jetson-stats`
2. reboot
3. Run jtop
4. See the following error
```
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
```

## Expected behavior

I want to be able to actually run jtop

## Additional context

I tried to reload service but it is not found. It does seems that service and setup config files have been installed in a wrong folder.

### Board

<details>
<summary><code>uname -a</code></summary>
<pre>
Linux jetson-nano-1-108 4.9.140-tegra #1 SMP PREEMPT Tue Oct 27 21:02:37 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
</pre>
</details>

<details>
<summary><code>jetson_release -v</code></summary>
<pre>
Software part of jetson-stats 4.2.3 - (c) 2023, Raffaello Bonghi
Model: NVIDIA Jetson Nano Developer Kit - Jetpack 4.4.1 [L4T 32.4.4]
NV Power Mode[1]: 5W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: p3448-0000
 - BoardIDs: p3448
 - Module: NVIDIA Jetson Nano (4 GB ram)
 - SoC: tegra210
 - CUDA Arch BIN: 5.3
 - Codename: Porg
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 18.04 Bionic Beaver
 - Release: 4.9.140-tegra
 - Python: 3.6.9
jtop:
 - Version: 4.2.3
 - Service: Active
Libraries:
 - CUDA: 10.2.89
 - cuDNN: 8.0.0.180
 - TensorRT: 7.1.3.0
 - VPI: 4.4.1
 - Vulkan: 1.2.70
 - OpenCV: 4.1.1 - with CUDA: NO
</pre>
</details>

### Log from jtop.service

<details>
[... 95 lines cut ...]
Nov 15 18:02:16 jetson-nano-1-108 jtop[12418]:     return _setlocale(category, locale)
Nov 15 18:02:16 jetson-nano-1-108 jtop[12418]: locale.Error: unsupported locale setting
Nov 15 18:02:17 jetson-nano-1-108 systemd[1]: jtop.service: Main process exited, code=exited, status=1/FAILURE
Nov 15 18:02:17 jetson-nano-1-108 systemd[1]: jtop.service: Failed with result 'exit-code'.
Nov 15 18:02:27 jetson-nano-1-108 systemd[1]: jtop.service: Service hold-off time over, scheduling restart.
Nov 15 18:02:27 jetson-nano-1-108 systemd[1]: jtop.service: Scheduled restart job, restart counter is at 1984.
Nov 15 18:02:27 jetson-nano-1-108 systemd[1]: Stopped jtop service.
Nov 15 18:02:27 jetson-nano-1-108 systemd[1]: Started jtop service.
</pre>
</details>

### Log from jetson-stats installation

<details>
<summary><code>sudo -H pip3 install --no-cache-dir -U jetson-stats</code></summary>
<pre>
Collecting jetson-stats
  Downloading https://files.pythonhosted.org/packages/83/7d/ee33e4ae4f02bc4b105f2efafde908ff98fe256b913093b4918802d7b7a5/jetson-stats-4.2.3.tar.gz (115kB)
    100% |████████████████████████████████| 122kB 6.8MB/s 
Requirement already up-to-date: distro in /usr/local/lib/python3.6/dist-packages (from jetson-stats)
Requirement already up-to-date: smbus2 in /usr/local/lib/python3.6/dist-packages (from jetson-stats)
Installing collected packages: jetson-stats
  Running setup.py install for jetson-stats ... done
</pre>
</details>

telemething, 2023-12-16:
also happening on Orin AGX

rbonghi, 2023-12-18:
Hi @halsimov I see where the bug come from, I will fix soon as possible

rbonghi, 2024-01-21:
Hi @mariobaldini 

Thank you for your post. I will fix this bug in the next release of Jetson-stats.

About nvpmodel I'll take a look soon

closed 2024-01-22 by rbonghi