rbonghi/jetson_stats #469: Error regarding memory usage display in Jtop 4.2.3

vesselofgod, 2023-12-02:
## Describe the bug

There's an error where the memory usage notation changes from 900Mib to 1100Mib (so if unit changes to 1.1 Gib) it's displayed as 100Mib.

## To Reproduce

It doesn't always happen, but it occurs when the memory usage unit changes on the Jetson.
(The above explanation is the most representative example)
The easiest way to reproduce the bug is to allocate resources without specifying a memory limit in the resources field.

[image]

## Expected behavior

In the example above, it is displayed as 100MiB when checked through ```jtop / free -h```, but when checking ```free -h``` in Linux, it is using 1.1GiB.

### Board

Output from `jetson_release -v`:

[image]

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -U jetson-stats`

[image]

artur-ag, 2023-12-20:
Seems to be the same as https://github.com/rbonghi/jetson_stats/issues/413.

rbonghi, 2024-01-05:
I am closing this issue.
Please respond directly to the other issue.

closed 2024-01-05 by rbonghi