rbonghi/jetson_stats #470: Jetpack missing [L4T 6.0]

jasl, 2023-12-06:
Please update jetson-stats with new jetpack
### Linux for Tegra

- L4T: JetPack 6.0 Developer Preview

### jetson-stats

- Version: 4.2.3

gehuageyan, 2023-12-06:
mark

qqice, 2023-12-06:
It seems that JetPack 6.0DP changed the way to access EMC freq again. I can't see EMC freq in tegrastats either. Maybe we should ask for help in [forums.developer.nvidia.com](https://forums.developer.nvidia.com)

qqice, 2023-12-06:
I've checked the tegrastats util and fixed the problem in #471.
Note: The fix is only for JP6.0 and won't work in JP5.1 before.

rbonghi, 2023-12-25:
Working in progress!
Thank you for your PR. I noticed some other bugs for this new Jetpack 6.0 DP version.
I'll release soon

rbonghi, 2023-12-27:
@jasl can you help me?

I merged your PR and others on `develop` branch, but I noticed something strange on the info page, can you send me a screenshot from page 7 (info page)

jasl, 2023-12-27:
> @jasl & @qqice can you help me?
> 
> I merged your PR and others on `develop` branch, but I noticed something strange on the info page, can you send me a screenshot from page 7 (info page)

Sure, I'm pulling dev branch, will reply you later

rbonghi, 2023-12-27:
thx! Looks like the new JP6.0 doesn't have enabled the I2C

jasl, 2023-12-27:
> @jasl & @qqice can you help me?
> 
> I merged your PR and others on `develop` branch, but I noticed something strange on the info page, can you send me a screenshot from page 7 (info page)

[image]

rbonghi, 2023-12-27:
I had the same experience. The I2C device is not detected from jtop, and the service doesn't recognize the board.
I also notice the SMBus get an error

[image]
`-bash: smbus2.smbus2.SMBus: No such file or directory`

I'm looking at the documentation to recognize the reason and fix it.

rbonghi, 2024-01-04:
released a new version **4.2.4**
```
sudo pip3 install -U jetson-stats
```
It's almost fixed. I noticed some bugs (not jtop related)
* It's missing the CPU model name (missing in `/proc/cpuinfo`)
* Missing `/proc/device-tree/nvidia,dtsfilename` and not available anymore the Codename variable

I close this issue, but please feel free to open a new one if you notice bugs

closed 2024-01-04 by rbonghi

Tavis991, 2024-01-23:
~/jetson_stats$ sudo systemctl status jtop.service
Unit jtop.service could not be found.

after installing from pip3 
is there a way to build from source ?

sbierly, 2024-01-23:
I am in Jetpack 5.1.2, just installed from SDK Manager. Running through my usual checklist, including jtop (why the heck isn't this installed standard by now?!).  I did:

```
sudo apt install python3-pip -y
sudo -H pip3 install -U jetson-stats
```
Rebooted, and then:

> scott@ubuntu:~$ jtop
> The jtop.service is not active. Please run:
> sudo systemctl restart jtop.service

> scott@ubuntu:~$ sudo systemctl restart jtop.service
> Failed to restart jtop.service: Unit jtop.service not found.

I saw a quite old thread with these same symptoms, which you appear to have addressed, yet here I am today with this again. Any ideas?

rbonghi, 2024-01-24:
Hi @sbierly 

thank you to write this message, I think the error is the same on this issue #488 

Can you please share me the output from 

```
sudo pip3 install --no-cache-dir -v -U jetson-stats
```

```
journalctl -u jtop.service -n 100 --no-pager
```

if the output from the latest command is empty can you please me share also the output from this command:
```
sudo JTOP_SERVICE=True jtop --force
```

sbierly, 2024-01-25:
scott@Orin4:~$ sudo pip3 install --no-cache-dir -v -U jetson-stats
[sudo] password for scott:
Non-user install because site-packages writeable
Created temporary directory: /tmp/pip-ephem-wheel-cache-ef3memzz
Created temporary directory: /tmp/pip-req-tracker-4wytwvjk
Initialized build tracking at /tmp/pip-req-tracker-4wytwvjk
Created build tracker: /tmp/pip-req-tracker-4wytwvjk
Entered build tracker: /tmp/pip-req-tracker-4wytwvjk
Created temporary directory: /tmp/pip-install-4ini5ebg
1 location(s) to search for versions of jetson-stats:
* https://pypi.org/simple/jetson-stats/
Fetching project page and analyzing links: https://pypi.org/simple/jetson-stats/
Getting page https://pypi.org/simple/jetson-stats/
Found index url https://pypi.org/simple
Getting credentials from keyring for https://pypi.org/simple
Getting credentials from keyring for pypi.org
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/jetson-stats/ HTTP/1.1" 200 7623
  Found link https://files.pythonhosted.org/packages/2d/d5/0db8f183b263300ac5a6f81c514b70ea8948c5f0db6d6a69ae4c30caeab3/jetson-stats-1.6.2.tar.gz#sha256=dee03f9b3a3cb3181e26ad54f72721460148169ee0f2db9eb97ad8e70e523688 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.2
  Found link https://files.pythonhosted.org/packages/ee/ef/f917a467a77762fea5aef431e96020e1677b5de202af621cc027a3b2d12e/jetson-stats-1.6.3.tar.gz#sha256=c6fef9e5cac1678c2f1e94431f34306080195c7d56485481776390f66ab25931 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.3
  Found link https://files.pythonhosted.org/packages/73/12/b6acd0f4fcd4427b9ff89b971f50da2cb0a25ced09192b5ea4aceb4e997c/jetson-stats-1.6.4.tar.gz#sha256=a15589fc8d5c2f43e1d1be36966f132a8e25adacef547a95c920107287fe7798 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.4
  Found link https://files.pythonhosted.org/packages/aa/f7/65931081ea159783c27933f1f41e2074067a23d0844e8904879e38b008db/jetson-stats-1.6.5.tar.gz#sha256=aec859a35d666fab9551853370dc2e34baa8b0ba9f397df688afe95db4e7a3f9 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.5
  Found link https://files.pythonhosted.org/packages/12/16/536096666ff809481b3b9ca85080e2011951412e75950b8d180c609c263d/jetson-stats-1.6.6.tar.gz#sha256=7789ebfdad42b10341d5c2aa3347f361d63e26f7a7a65a9c432a9fa9bbe1eb4b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.6
  Found link https://files.pythonhosted.org/packages/70/e2/8dd2dbffa2b60fa7fc3936515a455c2208a9880f52d8c69bba09ec5b989d/jetson-stats-1.6.7.tar.gz#sha256=38724beeecdb709b9d707ad75ad145e6852516d47d1c7a21c955fbdea48384f6 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.7
  Found link https://files.pythonhosted.org/packages/a8/58/fd74cee39ff63869b11e11d22a0548a8a801b6c44d4712b2644a26c6d991/jetson-stats-1.6.8.tar.gz#sha256=dd4c216fc8adc28b66147cee30609202ab5ee643b7aee43ba3c4352dea8bc51e (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.8
  Found link https://files.pythonhosted.org/packages/00/b8/b5f8f9422d3178b884eed48df32e562212972c37afff4485ff28d0df35db/jetson-stats-1.6.9.tar.gz#sha256=96cb05deb19a221adb581107a8f6fc00592fac2e2726fa24f43bafaea6916f70 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.9
  Found link https://files.pythonhosted.org/packages/7f/88/e777cb4e02e3470b88531b58ffcfcc86adb440a09dc9ad4d6987e36466a4/jetson-stats-1.6.10.tar.gz#sha256=0ec5c97e844d995f41deb768a3879be0ed7789c22e5be7f134c4bd9f2b9ed4b5 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.10
  Found link https://files.pythonhosted.org/packages/22/de/8068116eae57f94fb7bcab86e98ec3cc2161d135ac9d51c4a8acad0e14f9/jetson-stats-1.6.11.tar.gz#sha256=48937efbaa6710c5cc9988434385c9b7231d6ac83a76627288a94f3365c8d413 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.11
  Found link https://files.pythonhosted.org/packages/a3/11/994c5d324736118f45c4afcf0a4d366ab1f1a0cd117977d14171535c7e4a/jetson-stats-1.6.12.tar.gz#sha256=0baa9ae32db9d9de33847d6642f546b68df982b7d7fa06b6a62e190ebdd5664b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.12
  Found link https://files.pythonhosted.org/packages/42/1c/a54ce3e69422a1b627ac2e38a1b1a3d9e39f840ca5ed1d118d60c2e34aba/jetson-stats-1.6.13.tar.gz#sha256=fe81b7401ebf65388a521de56711e5108037d08366eea3497a2a39e347dd6d10 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.13
  Found link https://files.pythonhosted.org/packages/f4/c2/c70d35d7e94668bfc730cce675815130f49a7b2d01f9d162810708ab7d25/jetson-stats-1.7.0.tar.gz#sha256=c71547a8b8f629507aadc3b78b1df79ad0382963797f080ba3e1c3cf0f01894f (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.7.0
[... cut ...]
[INFO] jtop.core.power - Found name=1-00082 type=USB model=<EMPTY> in ucsi-source-psy-1-00082
[INFO] jtop.core.power - Found SYSTEM power monitor
[INFO] jtop.core.fan - Fan pwmfan(1) found in /sys/class/hwmon/hwmon2
[INFO] jtop.core.fan - RPM pwm_tach found in /sys/class/hwmon/hwmon0
[INFO] jtop.core.fan - Found nvfancontrol.service
[INFO] jtop.core.jetson_clocks - jetson_clocks found in /usr/bin/jetson_clocks
[INFO] jtop.core.nvpmodel - nvpmodel running in [2]MODE_30W - Default: 2
[ERROR] jtop.service - Group jtop does not exist!

rbonghi, 2024-01-25:
Can you test this release candidate for me? Looks fixed @sbierly @Tavis991

```
sudo -H pip3 install -v jetson-stats==4.2.6rc1
```

if you confirm I'll make a new version stable
Otherwise can you copy and paste the output of this installation

Thank you very much

rbonghi, 2024-01-25:
Fixed with new release **4.2.6**

Update with:
```
sudo -H pip3 install -U jetson-stats
```

sbierly, 2024-01-25:
Yes, confirming here as well, it's fixed thanks!
--Scott

Tavis991, 2024-01-25:
working ! great job buddy