rbonghi/jetson_stats #484: Directory missing on JP6

DYSpider13, 2024-01-08:
## Describe the bug

jetson-stats not working on JP6. Service is not starting due to some files (entire directory) missing: 

/sys/kernel/actmon_avg_activity/mc_all

Actually, it is not even detecting that Jetpack is installed. 

## To Reproduce

Steps to reproduce the behavior:

1. Try jetson-stats on JP6, kernel 5.15

## Screenshots

[image]

## Expected behavior

Shoud give the usual output of jtop

### Board

Output from `jetson_release -v`:

```
Software part of jetson-stats 4.2.4 - (c) 2024, Raffaello Bonghi
Jetpack missing!
 - Model: NVIDIA Jetson AGX Orin Developer Kit 
 - L4T: 36.0.0
NV Power Mode[2]: MODE_30W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13701-0004-500 F.0
 - P-Number: p3701-0004
 - Module: NVIDIA Jetson AGX Orin (32GB ram)
 - SoC: tegra234
 - CUDA Arch BIN: 8.7
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 22.04 Jammy Jellyfish
 - Release: 5.15.0-91-generic
 - Python: 3.10.12
jtop:
 - Version: 4.2.4
 - Service: Inactive
Libraries:
 - CUDA: Not installed
 - cuDNN: Not installed
 - TensorRT: Not installed
 - VPI: Not installed
 - OpenCV: 4.5.4 - with CUDA: NO
 
```

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

```
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.service - Running on Python: 3.10.12
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.cpu - Found 8 CPU
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [WARNING] jtop.core.gpu - No NVIDIA GPU available
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.processes - Process service started
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.memory - Found EMC!
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.memory - Memory service started
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
janv. 08 12:03:45 edge-seldon-jp6-id575 jtop[2586244]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
[... 79 lines cut ...]
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:   File "/usr/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:     self.run()
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:   File "/usr/local/lib/python3.10/dist-packages/jtop/service.py", line 310, in run
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:     data = self.jtop_decode()
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:   File "/usr/local/lib/python3.10/dist-packages/jtop/service.py", line 575, in jtop_decode
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:     data['mem'] = self.memory.get_status(total)
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:   File "/usr/local/lib/python3.10/dist-packages/jtop/core/memory.py", line 431, in get_status
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:     utilization = int(cat(self._root_path + "/actmon_avg_activity/mc_all"))
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:   File "/usr/local/lib/python3.10/dist-packages/jtop/core/common.py", line 109, in cat
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]:     with open(path, 'r') as f:
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589161]: FileNotFoundError: [Errno 2] No such file or directory: '/sys/kernel/actmon_avg_activity/mc_all'
janv. 08 12:07:19 edge-seldon-jp6-id575 jtop[2589139]: [INFO] jtop.service - Service closed
janv. 08 12:07:19 edge-seldon-jp6-id575 systemd[1]: jtop.service: Deactivated successfully.

```

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -U jetson-stats`

```
Requirement already satisfied: jetson-stats in /usr/local/lib/python3.10/dist-packages (4.2.4)
Requirement already satisfied: smbus2 in /usr/local/lib/python3.10/dist-packages (from jetson-stats) (0.4.3)
Requirement already satisfied: distro in /usr/lib/python3/dist-packages (from jetson-stats) (1.7.0)
```

rbonghi, 2024-01-08:
Hi @DYSpider13 

How did you install the Jetpack 6.0? Looking at your log, you have a storage L4T=36.0.0 when should be [L4T=36.2.0](https://developer.nvidia.com/embedded/jetson-linux-r362)

Can you please share the output from:
```
jtop --error-log
```

Best,
Raffaello

DYSpider13, 2024-01-08:
Hi @rbonghi , 

It came from the vendor himself. 

Here is the output of jtop --error-log

```
--------------------- PLATFORM -------------------------
Machine: aarch64
System: Linux
Distribution: Ubuntu 22.04 Jammy Jellyfish
Release: 5.15.0-91-generic
Python: 3.10.12
-------------------- RAW OUTPUT ------------------------
------------------
/etc/nv_tegra_release:
# R36 (release), REVISION: 0.0, GCID: 34507387, BOARD: generic, EABI: aarch64, DATE: Wed Oct  4 16:41:43 UTC 2023
------------------
/sys/firmware/devicetree/base/model:
NVIDIA Jetson AGX Orin Developer Kit Axiomtek
------------------
/proc/device-tree/nvidia,boardids:
No such file or directory
------------------
/proc/device-tree/compatible:
nvidia,p3737-0000+p3701-0004nvidia,p3701-0004nvidia,tegra234
------------------
/proc/device-tree/nvidia,dtsfilename:
No such file or directory
------------------
I2C-0-0x50:
02 00 FE 00 00 00 00 00 00 00 00 00 00 00 00 00    ..þ.............
00 00 00 0A 36 39 39 2D 31 33 37 30 31 2D 30 30    ....699-13701-00
30 34 2D 35 30 30 20 46 2E 30 00 00 00 00 00 00    04-500 F.0......
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 B7 0C 95 2D B0 48 31 34 32 32 38 32    ....·..-°H142282
32 30 36 31 36 33 33 00 00 00 00 00 00 00 00 00    2061633.........
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 4E 56 43 42 00 00 4D 31 00 00    ......NVCB..M1..
00 00 00 00 00 00 00 00 00 00 00 00 B7 0C 95 2D    ............·..-
B0 48 0A 00 00 00 00 00 00 00 00 00 00 00 00 00    °H..............
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 81    ................

------------------
I2C-0-0x56:
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF    ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ

------------------
I2C-1:
FAIL
------------------
I2C-2:
FAIL
------------------
I2C-7:
FAIL

Log from jtop 4.2.4

```

rbonghi, 2024-01-08:
I'm figuring out which type of Jetpack you have; It's something else compared to the official NVIDIA release.
I also noticed another issue: the GPU is not detected, and I would like to figure out where is locate the GPU.

Can you share the output from this script?
This script (I hope) helps me to understand and fix jtop :-)

```python
import os
igpu_path = "/sys/class/devfreq/"

for item in os.listdir(igpu_path):
    item_path = os.path.join(igpu_path, item)
    if os.path.isfile(item_path) or os.path.islink(item_path):
        # Check name device
        name_path = "{item}/device/of_node/name".format(item=item_path)
        if os.path.isfile(name_path):
            # Decode name
            with open(name_path, 'r') as f:
                name = f.readline().rstrip('\x00')
            # path and file
            print("Path: {}".format(name_path))
            print("{}".format(name))
```

Thank you in advance

DYSpider13, 2024-01-08:
@rbonghi 

It seems that it is a special release from the vendor. 

The script does not return anything, since the "/sys/class/devfreq/" is empty.

rbonghi, 2024-01-08:
Thank you!
I was with the same feeling. Looks like some devices are missing or located in some folders that I don't know

Have you tried to run:
```
sudo tegrastats
```
If there is an output, can you share here?

Thank you again

rbonghi, 2024-01-08:
There might be something strange with your flashed Jetson.

I suggest working with your vendor to reflash your device to the official [Jetpack 6.0 DP](https://developer.nvidia.com/embedded/jetpack-sdk-60dp), while I figure out how to fix other bugs.

DYSpider13, 2024-01-08:
That's my feeling too. I'll check with him. Thanks @rbonghi !

rbonghi, 2024-01-09:
I close this issue, if you've again bug on your device with jtop feel free to reopen it.

closed 2024-01-09 by rbonghi