rbonghi/jetson_stats #488: Jtop not installing correctly in L4T 5.1.2 on AGX Orin Dev Kit 32GB

sbierly, 2024-01-24:
## Describe the bug

This has failed consistently on 4 cleanly built systems. Used SDK manager to "flash" the OS, then used desktop to connect wifi, and then remotely (ssh) installed Jetpack using 

sudo apt update
sudo apt upgrade
sudo apt install nvidia-jetpack

First thing after that is to install Jetson Stats:

sudo apt install python3-pip -y
sudo pip3 install -U jetson-stats
sudo systemctl restart jetson_stats.service
sudo reboot now

jtop

```
scott@Orin4:~$ jtop
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
scott@Orin4:~$ sudo systemctl restart jtop.service
Failed to restart jtop.service: Unit jtop.service not found.
scott@Orin4:~$
```

## To Reproduce

Steps to reproduce the behavior:

See above

## Screenshots

If applicable, add screenshots to help explain your problem.

## Expected behavior

A clear and concise description of what you expected to happen.

## Additional context

Add any other context about the problem here.

### Board

Output from `jetson_release -v`:
```
scott@Orin4:~$ jetson_release -v
Software part of jetson-stats 4.2.5 - (c) 2024, Raffaello Bonghi
Model: Jetson AGX Orin Developer Kit - Jetpack 5.1.2 [L4T 35.4.1]
NV Power Mode[2]: MODE_30W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13701-0000-500 J.0
 - P-Number: p3701-0000
 - Module: NVIDIA Jetson AGX Orin
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: Concord
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.120-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.5
 - Service: Inactive
Libraries:
[... 189 lines cut ...]
------------------
Path: I2C-1
FAIL
------------------
Path: I2C-2
FAIL
------------------
Path: I2C-7
FAIL

-------------------- IGPU OUTPUT ---------------------
------------------
Path: /sys/class/devfreq/17000000.ga10b/device/of_node/name
ga10b
------------------
Path: /sys/class/devfreq/15480000.nvdec/device/of_node/name
nvdec
------------------
Path: /sys/class/devfreq/154c0000.nvenc/device/of_node/name
nvenc
------------------
Path: /sys/class/devfreq/15340000.vic/device/of_node/name
vic

Log from jtop 4.2.5

649519404, 2024-01-24:
Xavier_ NX 5.1.2 also has the same issue, how should we handle it

rbonghi, 2024-01-24:
Hi @sbierly  and @649519404 
Thank you in advance for your open issue.

Can you please share the output when you run this command?
This will help me to understand where this bug is coming from
```
sudo JTOP_SERVICE=True jtop --force
```

haimat, 2024-01-24:
Hi, I have the same issue:

```
$ jtop
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service

$ sudo systemctl restart jtop.service
Failed to restart jtop.service: Unit jtop.service not found.

$ jetson_release 
Software part of jetson-stats 4.2.5 - (c) 2024, Raffaello Bonghi
Model: ThinkEdge SE70 (NVIDIA Jetson Xavier NX) - Jetpack 5.0.2 GA [L4T 35.1.0]
NV Power Mode[8]: MODE_20W_6CORE
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: p3668-0001
 - Module: NVIDIA Jetson Xavier NX
Platform:
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.104-tegra
jtop:
 - Version: 4.2.5
 - Service: Inactive
Libraries:
 - CUDA: Not installed
 - cuDNN: Not installed
 - TensorRT: Not installed
 - VPI: Not installed
 - Vulkan: 1.3.203
 - OpenCV: 4.5.4 - with CUDA: NO

$ JTOP_SERVICE=True sudo jtop --force
Traceback (most recent call last):
  File "/home/sinntelligence/.local/bin/jtop", line 5, in <module>
    from jtop.__main__ import main
ModuleNotFoundError: No module named 'jtop'
```

rbonghi, 2024-01-24:
Hi @haimat 

Can you please share the output from

```
sudo -H pip3 install --no-cache-dir -v -U jetson-stats
```

haimat, 2024-01-24:
```
$ sudo -H pip3 install --no-cache-dir -v -U jetson-stats
Non-user install because site-packages writeable
Created temporary directory: /tmp/pip-ephem-wheel-cache-787c4kw_
Created temporary directory: /tmp/pip-req-tracker-onie4o7c
Initialized build tracking at /tmp/pip-req-tracker-onie4o7c
Created build tracker: /tmp/pip-req-tracker-onie4o7c
Entered build tracker: /tmp/pip-req-tracker-onie4o7c
Created temporary directory: /tmp/pip-install-1c2iek65
1 location(s) to search for versions of jetson-stats:
* https://pypi.org/simple/jetson-stats/
Fetching project page and analyzing links: https://pypi.org/simple/jetson-stats/
Getting page https://pypi.org/simple/jetson-stats/
Found index url https://pypi.org/simple
Getting credentials from keyring for https://pypi.org/simple
Getting credentials from keyring for pypi.org
Starting new HTTPS connection (1): pypi.org:443
https://pypi.org:443 "GET /simple/jetson-stats/ HTTP/1.1" 200 7623
  Found link https://files.pythonhosted.org/packages/2d/d5/0db8f183b263300ac5a6f81c514b70ea8948c5f0db6d6a69ae4c30caeab3/jetson-stats-1.6.2.tar.gz#sha256=dee03f9b3a3cb3181e26ad54f72721460148169ee0f2db9eb97ad8e70e523688 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.2
  Found link https://files.pythonhosted.org/packages/ee/ef/f917a467a77762fea5aef431e96020e1677b5de202af621cc027a3b2d12e/jetson-stats-1.6.3.tar.gz#sha256=c6fef9e5cac1678c2f1e94431f34306080195c7d56485481776390f66ab25931 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.3
  Found link https://files.pythonhosted.org/packages/73/12/b6acd0f4fcd4427b9ff89b971f50da2cb0a25ced09192b5ea4aceb4e997c/jetson-stats-1.6.4.tar.gz#sha256=a15589fc8d5c2f43e1d1be36966f132a8e25adacef547a95c920107287fe7798 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.4
  Found link https://files.pythonhosted.org/packages/aa/f7/65931081ea159783c27933f1f41e2074067a23d0844e8904879e38b008db/jetson-stats-1.6.5.tar.gz#sha256=aec859a35d666fab9551853370dc2e34baa8b0ba9f397df688afe95db4e7a3f9 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.5
  Found link https://files.pythonhosted.org/packages/12/16/536096666ff809481b3b9ca85080e2011951412e75950b8d180c609c263d/jetson-stats-1.6.6.tar.gz#sha256=7789ebfdad42b10341d5c2aa3347f361d63e26f7a7a65a9c432a9fa9bbe1eb4b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.6
  Found link https://files.pythonhosted.org/packages/70/e2/8dd2dbffa2b60fa7fc3936515a455c2208a9880f52d8c69bba09ec5b989d/jetson-stats-1.6.7.tar.gz#sha256=38724beeecdb709b9d707ad75ad145e6852516d47d1c7a21c955fbdea48384f6 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.7
  Found link https://files.pythonhosted.org/packages/a8/58/fd74cee39ff63869b11e11d22a0548a8a801b6c44d4712b2644a26c6d991/jetson-stats-1.6.8.tar.gz#sha256=dd4c216fc8adc28b66147cee30609202ab5ee643b7aee43ba3c4352dea8bc51e (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.8
  Found link https://files.pythonhosted.org/packages/00/b8/b5f8f9422d3178b884eed48df32e562212972c37afff4485ff28d0df35db/jetson-stats-1.6.9.tar.gz#sha256=96cb05deb19a221adb581107a8f6fc00592fac2e2726fa24f43bafaea6916f70 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.9
  Found link https://files.pythonhosted.org/packages/7f/88/e777cb4e02e3470b88531b58ffcfcc86adb440a09dc9ad4d6987e36466a4/jetson-stats-1.6.10.tar.gz#sha256=0ec5c97e844d995f41deb768a3879be0ed7789c22e5be7f134c4bd9f2b9ed4b5 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.10
  Found link https://files.pythonhosted.org/packages/22/de/8068116eae57f94fb7bcab86e98ec3cc2161d135ac9d51c4a8acad0e14f9/jetson-stats-1.6.11.tar.gz#sha256=48937efbaa6710c5cc9988434385c9b7231d6ac83a76627288a94f3365c8d413 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.11
  Found link https://files.pythonhosted.org/packages/a3/11/994c5d324736118f45c4afcf0a4d366ab1f1a0cd117977d14171535c7e4a/jetson-stats-1.6.12.tar.gz#sha256=0baa9ae32db9d9de33847d6642f546b68df982b7d7fa06b6a62e190ebdd5664b (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.12
  Found link https://files.pythonhosted.org/packages/42/1c/a54ce3e69422a1b627ac2e38a1b1a3d9e39f840ca5ed1d118d60c2e34aba/jetson-stats-1.6.13.tar.gz#sha256=fe81b7401ebf65388a521de56711e5108037d08366eea3497a2a39e347dd6d10 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.6.13
  Found link https://files.pythonhosted.org/packages/f4/c2/c70d35d7e94668bfc730cce675815130f49a7b2d01f9d162810708ab7d25/jetson-stats-1.7.0.tar.gz#sha256=c71547a8b8f629507aadc3b78b1df79ad0382963797f080ba3e1c3cf0f01894f (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 1.7.0
[... cut ...] 0, 4.1.1, 4.1.2, 4.1.3, 4.1.4, 4.1.5, 4.2.0, 4.2.1, 4.2.2, 4.2.3, 4.2.4, 4.2.5)
Requirement already up-to-date: jetson-stats in /usr/local/lib/python3.8/dist-packages (4.2.5)
Requirement already satisfied, skipping upgrade: smbus2 in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (0.4.3)
Requirement already satisfied, skipping upgrade: distro in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (1.9.0)
Cleaning up...
Removed build tracker: '/tmp/pip-req-tracker-onie4o7c'
```

rbonghi, 2024-01-24:
I'm investigating where this bug is coming from.
I keep you posted

rbonghi, 2024-01-24:
Reference to this comment https://github.com/rbonghi/jetson_stats/issues/366#issuecomment-1908030304
I'll fix in a day with a new release

sbierly, 2024-01-24:
Hi Raffaello, thanks for the quick response! I simply love jtop, and don't
get why Nvidia doesn't care to just include it in the release, they provide
nothing like this and it's essential! Here you go:

***@***.***:~$ sudo JTOP_SERVICE=True jtop --force
[sudo] password for scott:
[INFO] jtop.core.config - Build service folder in /usr/local/jtop
[INFO] jtop.service - jetson_stats 4.2.5 - server loaded
[INFO] jtop.service - Running on Python: 3.8.10
[INFO] jtop.core.hardware - Hardware detected aarch64
[INFO] jtop.core.hardware - NVIDIA Jetson 699-level Part
Number=699-13701-0000-500 J.0
[INFO] jtop.core.hardware - NVIDIA Jetson Module=NVIDIA Jetson AGX Orin
[INFO] jtop.core.hardware - NVIDIA Jetson detected L4T=35.4.1
[INFO] jtop.core.cpu - Found 12 CPU
[INFO] jtop.core.gpu - GPU "ga10b" status in
/sys/devices/platform/17000000.ga10b
[INFO] jtop.core.gpu - GPU "ga10b" frq in
/sys/devices/platform/17000000.ga10b/devfreq/17000000.ga10b
[INFO] jtop.core.processes - Process service started
[INFO] jtop.core.memory - Found EMC!
[INFO] jtop.core.memory - Memory service started
[INFO] jtop.core.engine - Special Engine group found: [dlaX]
[INFO] jtop.core.engine - Special Engine group found: [pvaX]
[INFO] jtop.core.engine - Engines found: [APE DLA0 DLA1 NVDEC NVENC NVJPG
OFA PVA0 SE VIC]
[INFO] jtop.core.temperature - Found thermal "CV0" in thermal_zone2
[INFO] jtop.core.temperature - Found thermal "CPU" in thermal_zone0
[INFO] jtop.core.temperature - Found thermal "Tboard" in thermal_zone9
[INFO] jtop.core.temperature - Found thermal "SOC2" in thermal_zone7
[INFO] jtop.core.temperature - Found thermal "Tdiode" in thermal_zone10
[INFO] jtop.core.temperature - Found thermal "SOC0" in thermal_zone5
[INFO] jtop.core.temperature - Found thermal "CV1" in thermal_zone3
[INFO] jtop.core.temperature - Found thermal "GPU" in thermal_zone1
[INFO] jtop.core.temperature - Found thermal "tj" in thermal_zone8
[INFO] jtop.core.temperature - Found thermal "SOC1" in thermal_zone6
[INFO] jtop.core.temperature - Found thermal "CV2" in thermal_zone4
[WARNING] jtop.core.power - Skipped NC
/sys/bus/i2c/devices/1-0041/hwmon/hwmon4/in1_label
[INFO] jtop.core.power - Alarms VDDQ_VDD2_1V8AO - {'crit_alarm': 0,
'max_alarm': 0}
[WARNING] jtop.core.power - Skipped NC
/sys/bus/i2c/devices/1-0041/hwmon/hwmon4/in3_label
[WARNING] jtop.core.power - Skipped "sum of shunt voltages"
/sys/bus/i2c/devices/1-0041/hwmon/hwmon4/in7_label
[INFO] jtop.core.power - Alarms VDD_GPU_SOC - {'crit_alarm': 0,
'max_alarm': 0}
[INFO] jtop.core.power - Alarms VDD_CPU_CV - {'crit_alarm': 0, 'max_alarm':
0}
[INFO] jtop.core.power - Alarms VIN_SYS_5V0 - {'crit_alarm': 0,
'max_alarm': 0}
[WARNING] jtop.core.power - Skipped "sum of shunt voltages"
/sys/bus/i2c/devices/1-0040/hwmon/hwmon3/in7_label
[INFO] jtop.core.power - Found I2C power monitor
[INFO] jtop.core.power - Found name=1-00081 type=USB model=<EMPTY> in
ucsi-source-psy-1-00081
[INFO] jtop.core.power - Found name=1-00082 type=USB model=<EMPTY> in
ucsi-source-psy-1-00082
[INFO] jtop.core.power - Found SYSTEM power monitor
[INFO] jtop.core.fan - Fan pwmfan(1) found in /sys/class/hwmon/hwmon2
[INFO] jtop.core.fan - RPM pwm_tach found in /sys/class/hwmon/hwmon0
[INFO] jtop.core.fan - Found nvfancontrol.service
[INFO] jtop.core.jetson_clocks - jetson_clocks found in
/usr/bin/jetson_clocks
[INFO] jtop.core.nvpmodel - nvpmodel running in [2]MODE_30W - Default: 2
[ERROR] jtop.service - Group jtop does not exist!

--Scott

razvanphp, 2024-01-25:
Indeed, seems to be the docker detection, I've downgraded to version `4.2.4` and there the `jtop.service` was correctly installed. Thank you for this great piece of software!

rbonghi, 2024-01-25:
Can you test this release candidate for me, looks fixed @razvanphp @haimat @sbierly @649519404 

```
sudo -H pip3 install -v jetson-stats==4.2.6rc1
```

if you confirm I'll make a new version stable

haimat, 2024-01-25:
@rbonghi Yes, with

```console
sudo -H pip3 install -v jetson-stats==4.2.6rc1
```

it works fine :+1: 
Thanks for your quick fix!

rbonghi, 2024-01-25:
Great, I'm making a new release 4.2.6 now

rbonghi, 2024-01-25:
Fixed with new release **4.2.6**

Update with:
```
sudo -H pip3 install -U jetson-stats
```

closed 2024-01-25 by rbonghi

sbierly, 2024-01-25:
It works! Fantastic, thank you so much for your fantastic contribution to
Jetson users worldwide!

And wait, more surprises, you added TOTAL Power, I was actually planning on
submitting a feature request, and there it is now! I have been away from
these projects for nearly a year, so missed that you had added this, more
thanks!
--Scott

B4njoS, 2024-05-01:
Hey @rbonghi !
The command `sudo JTOP_SERVICE=True jtop --force` worked for me when I have opened 2 terminals, one to keep the command alive and the other one to check and the service was really running.
I'm using version 4.2.7 on Orin NX Jetpack 5.1

$ jetson_release -v

Software part of jetson-stats 4.2.7 - (c) 2024, Raffaello Bonghi
Model: NVIDIA Orin NX Developer Kit - Jetpack 5.1 [L4T 35.2.1]
NV Power Mode[3]: 25W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - Module: Check with sudo
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: P3768
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 Focal Fossa
 - Release: 5.10.104-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.7
 - Service: Inactive
Libraries:
 - CUDA: 11.4.315
 - cuDNN: 8.6.0.166
 - TensorRT: 8.5.2.2
 - VPI: 2.2.4
 - Vulkan: 1.3.204
 - OpenCV: 4.5.4 - with CUDA: NO

$ sudo jtop
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
$ sudo systemctl restart jtop.service
Failed to restart jtop.service: Unit jtop.service not found.

$ jtop --version
jtop 4.2.7

I have tried with many other versions, only the solution you gave above helped me fix my issue but it's not a solution because the command needs to be running all the time before running `jtop`
Hope to fix it by myself and not interrupt you, I'll update here too.. Thanks!

B4njoS, 2024-05-01:
I managed to fix the problem with some reboots and install and uninstall
But now it alerts me about missing files

-bash: /usr/local/lib/python3.8/dist-packages/jtop/jetson_variables: No such file or directory
-bash: /usr/local/lib/python3.8/dist-packages/jtop/jetson_libraries: No such file or directory

Is this something to be afraid of?