rbonghi/jetson_stats #495: [WARN] jetson-stats not supported for [L4T 35.5.0]

Jen-Hung-Ho, 2024-02-23:
## Describe the bug
Jetop not detect jetpack version 5.1.3 on Jetson Orion Nano

A clear and concise description of what the bug is.

## To Reproduce
Install Jetpack version 5.1.3 

1. run sudo jtop.  [see screenshost]
2. then quit jtop  
3. jetbot@ubuntu:~$ sudo jtop
[WARN] jetson-stats not supported for [L4T 35.5.0]
  Please, try: sudo pip3 install -U jetson-stats or
  open a Github issue (press CTRL + Click)

jetbot@ubuntu:~$ apt-cache show nvidia-jetpack
Package: nvidia-jetpack
Version: 5.1.3-b29
Architecture: arm64
Maintainer: NVIDIA Corporation
Steps to reproduce the behavior:

## Screenshots
[image]
[image]

Jen-Hung-Ho, 2024-02-23:
Reference 
[Jetpack Not Detected Or Missing On Jetson Orin Nano](https://forums.developer.nvidia.com/t/jetpack-not-detected-or-missing-on-jetson-orin-nano/263150)
Down grade to JetPack 5.1.2.  L4t: 35.4.3 fix the issue.

closed 2024-02-23 by Jen-Hung-Ho

rbonghi, 2024-03-04:
This bug will be fixed later today

rbonghi, 2024-03-18:
Fixed with the new release. Please update to **4.2.7**

```
sudo pip3 install -U jetson-stats
```