rbonghi/jetson_stats #497: jetson-stats not supported for [L4T 35.5.0]

rsalas-thecne, 2024-02-27:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 35.5.0

### Jetson-Stats

 - Version: 4.2.6

En Jetson orin nano, el comando jtop no encuentra la version del jetpack "Jetpack: Missing", al ejecutar el comando sudo apt show nvidia-jetpack si reconoce jetpack!

Dapid, 2024-03-01:
Translation:

In Jetson Orin Nano, jtop canot find jetpack, reporting it as missing, although `apt show nvidia-jetpack` does show it is installed.

I can reproduce on my end.

nvtom, 2024-03-04:
"36.2.0": "6.0 DP",
    # -------- JP5 -------- 
    # to fix you may add following line with jetpack 35.5.0 definition, try and install from a local copy with sudo pip3 install -e .  Note that this is a quick fix and there may be other adaptations needed to support the new jetpack release fully 
    "35.5.0": "5.1.3",
    "35.4.1": "5.1.2",

nvtom, 2024-03-04:
PS: this is line #47 in /jtop/core/jetson_variables.py

rbonghi, 2024-03-04:
Hi all,

I fix this bug for today.
Sorry for the delay

rsalas-thecne, 2024-03-04:
thanks, problem solved.
[image]

rbonghi, 2024-03-18:
Fixed with the new release. Please update to **4.2.7**

```
sudo pip3 install -U jetson-stats
```