rbonghi/jetson_stats #509: vulkan indicator wrong

theofficialgman, 2024-03-23:
you parse incorrect output from vulkaninfo leading to incorrect output

you should report back the vulkan apiVersion
eg:
```
Device Properties and Extensions:
=================================
GPU0:
VkPhysicalDeviceProperties:
---------------------------
    apiVersion        = 4198485 (1.1.85)
    driverVersion     = 134266944 (0x800c040)
    vendorID          = 0x10de
    deviceID          = 0x92ba03d7
    deviceType        = PHYSICAL_DEVICE_TYPE_INTEGRATED_GPU
    deviceName        = NVIDIA Tegra X1 (nvgpu)
    pipelineCacheUUID = 9f41a1f6-6b2d-37a1-83f3-32e8a578d389
```
but instead you report back the instance (loader) version
```
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.204
```
the loader can be updated independent from the driver. loader comes from the distro.

rbonghi, 2024-05-07:
Hi @theofficialgman 
Thank you for this message and apologize for my late reply.

Thank you to let me notice on this mismatch.
Looking the NVIDIA documentation https://developer.nvidia.com/embedded/jetpack 
I think should be highlighted the Instance version and the SC version.

I take a look this bug and fix for the next release

theofficialgman, 2024-05-08:
> I think should be highlighted the Instance version and the SC version.

No, I already explained that. Instance (loader) comes from distro repositories. It can be much higher than the driver supported Vulkan API version. You need to report the `apiVersion` for the appropriate hardware adapter like I have shown in my first post.

I do not know where the Vulkan SC version is reported as I don't use a device which supports that.

mrcmunir, 2024-06-13:
> Hi @theofficialgman Thank you for this message and apologize for my late reply.
>
> Thank you to let me notice on this mismatch. Looking the NVIDIA documentation [...]

The problem it's in here https://github.com/rbonghi/jetson_stats/blob/58a275ee3fa4ffc9e12f30ab7c19ceadb372f1a8/jtop/core/jetson_libraries.py#L124
You search Vulkan Instance Version and not "apiVersion        ="

closed 2026-10-15 by johnnynunez