rbonghi/jetson_stats #510: Jtop Service Crashes on Jetson Xavier When CPU Usage Exceeds 90%

ozknsmz, 2024-03-26:
## Describe the bug

When running the jtop service on Jetson devices, the service crashes when the CPU value of the Jetson Xavier exceeds 90%.

## To Reproduce

Steps to reproduce the behavior:

1. Start the jtop service on a Jetson device.
2. Monitor the CPU usage using jtop.
3. Increase the CPU load until it exceeds 90%.

## Expected behavior

The jtop service should continue running and displaying jetson stats informations even when the CPU load is high.

### Board

Output from `jetson_release -v`:

* jetson-stats version: 4.2.6
* Module: NVIDIA Jetson Xavier NX  8GB
* Jetpack: 5.1.1
* L4T: 35.3.1

### Log from jtop.service
The jetson_stats.service is not active. Please run: sudo systemctl restart jetson_stats.service

pimientofrito, 2024-05-27:
Hi! Did you manage to resolve this? I have the same problem on Orin

ozknsmz, 2024-05-27:
> Hi! Did you manage to resolve this? I have the same problem on Orin

The issue has not been resolved yet, but they were aware of the issue and stated they would work on it.