rbonghi/jetson_stats #512: Constant "jtop.service is not active" error on Jetson nano

TakShimoda, 2024-04-01:
## Describe the bug

I'm using a Jetson nano with jetpack 4.6.2. I used the normal command `sudo pip3 install -U jetson-stats` to install on two other Jetson nanos and they work fine, but for one of them, I'm stuck with errors. 

On the first time installing, the command `jtop` did nothing (after I typed it, nothing showed on the screen, not even errors).

I couldn't find any issues related to mine, so I just uninstalling and re-installing, but after that, `sudo pip3 install -U jetson-stats` gave an error "zipfile.badzipfile: file is not a zip file".

I found an alternative way to install [here](https://github.com/rbonghi/jetson_stats/issues/86#issuecomment-722288802), which installed, but every time I run `jtop` I get the following error message:

```
Skip for setup.py
The jtop.service is not active. Please run:
sudo systemctl restart jtop.service
```
I tried `sudo systemctl restart jtop.service` but it keeps repeating the same error.
As shown below, it seems `NameError: name 'SMBus' is not defined` is the error (it even shows this error when trying the `jetson_release -v` command). However, when I tried `sudo pip3 install smbus2` I get the following:

   `Requirement already satisfied: smbus2 in /usr/local/lib/python3.8/dist-packages (0.4.3)`

I also tried `sudo apt-get install python3-smbus` but it didn't solve any errors.

### Board

Output from `jetson_release -v`:
```
Skip for setup.py
Software part of jetson-stats 4.2.7 - (c) 2024, Raffaello Bonghi
Traceback (most recent call last):
  File "/usr/local/bin/jetson_release", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/jetson_release.py", line 45, in main
    jetson = get_jetson_variables()
  File "/usr/local/lib/python3.8/dist-packages/jtop/core/jetson_variables.py", line 303, in get_jetson_variables
    part_number, jetson_part_number = get_part_number()
  File "/usr/local/lib/python3.8/dist-packages/jtop/core/jetson_variables.py", line 279, in get_part_number
    bus = SMBus(bus_number, force=True)
NameError: name 'SMBus' is not defined
```

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

```
-- Logs begin at Tue 2023-11-21 22:10:21 CET, end at Mon 2024-04-01 23:03:14 CEST. --
Apr 01 23:02:37 nano jtop[9139]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/hardware.py", line 71, in get_hardware
Apr 01 23:02:37 nano jtop[9139]:     jetson = get_jetson_variables()
Apr 01 23:02:37 nano jtop[9139]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/jetson_variables.py", line 303, in get_jetson_variables
Apr 01 23:02:37 nano jtop[9139]:     part_number, jetson_part_number = get_part_number()
Apr 01 23:02:37 nano jtop[9139]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/jetson_variables.py", line 279, in get_part_number
Apr 01 23:02:37 nano jtop[9139]:     bus = SMBus(bus_number, force=True)
Apr 01 23:02:37 nano jtop[9139]: NameError: name 'SMBus' is not defined
Apr 01 23:02:37 nano systemd[1]: jtop.service: Main process exited, code=exited, status=1/FAILURE
Apr 01 23:02:37 nano systemd[1]: jtop.service: Failed with result 'exit-code'.
Apr 01 23:02:40 nano systemd[1]: Stopped jtop service.
Apr 01 23:02:40 nano systemd[1]: Started jtop service.
Apr 01 23:02:40 nano jtop[9178]: [INFO] jtop.service - jetson_stats 4.2.7 - server loaded
Apr 01 23:02:40 nano jtop[9178]: [INFO] jtop.service - Running on Python: 3.8.10
Apr 01 23:02:40 nano jtop[9178]: [INFO] jtop.core.hardware - Hardware detected aarch64
Apr 01 23:02:40 nano jtop[9178]: Skip for setup.py
Apr 01 23:02:41 nano jtop[9178]: Traceback (most recent call last):
Apr 01 23:02:41 nano jtop[9178]:   File "/usr/local/bin/jtop", line 8, in <module>
Apr 01 23:02:41 nano jtop[9178]:     sys.exit(main())
Apr 01 23:02:41 nano jtop[9178]:   File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 104, in main
Apr 01 23:02:41 nano jtop[9178]:     server = JtopServer(force=args.force)
Apr 01 23:02:41 nano jtop[9178]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 281, in __init__
Apr 01 23:02:41 nano jtop[9178]:     self.board = {'hardware': get_hardware()}
Apr 01 23:02:41 nano jtop[9178]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/hardware.py", line 71, in get_hardware
Apr 01 23:02:41 nano jtop[9178]:     jetson = get_jetson_variables()
[... 172 lines cut ...]
  Found link https://files.pythonhosted.org/packages/2f/a0/1b0ce82067425748cc90b1f3c67349dc7b733d499c47dd6ab3a0bf5a7467/jetson-stats-4.2.0rc1.tar.gz#sha256=7de61ef0de9de8c0383b1486bdf0065431d8d3f5ec88d2b47de67277b5c0d6bf (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 4.2.0rc1
  Found link https://files.pythonhosted.org/packages/f5/77/2c8c7105b94449237e53b455f4731ce3213e341a796085630e31ac4f6e8c/jetson-stats-4.2.0rc2.tar.gz#sha256=3d109c67ea59f7cb81411c92ac222ce1747d14c10a606f6a95aee618fa28dec0 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 4.2.0rc2
  Found link https://files.pythonhosted.org/packages/84/af/4622fde4d0cb6f43fdf58b5f45d4714c73d27c510618f5e31096b45ae2fc/jetson-stats-4.2.0.tar.gz#sha256=cd09a25590407ed8024fb68daab5fdabeebfd0dc173c3b108994de3968104504 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 4.2.0
  Found link https://files.pythonhosted.org/packages/aa/07/098bfb6f864b44c12957be8798c34af4faabac33ffde2eaf1ef861f901e5/jetson-stats-4.2.1.tar.gz#sha256=861144a0babdd36b688373e7dac8c0373a0e378c1ae9c19ccacceec4ce270b51 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, <4), version: 4.2.1
  Found link https://files.pythonhosted.org/packages/db/e3/74ddedecf2210240cf4cc7f7509f5a1804fcded604cb5ee0249ffb485722/jetson-stats-4.2.2.tar.gz#sha256=6b2434e8d4a9f0d7b06742f312ddba4bee042228ee0f188eaa49297ca5fd91cb (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.2
  Found link https://files.pythonhosted.org/packages/83/7d/ee33e4ae4f02bc4b105f2efafde908ff98fe256b913093b4918802d7b7a5/jetson-stats-4.2.3.tar.gz#sha256=0e4b347583a68406919b3d7fc5509db1ff785ae97839cb1e0f72e9460c977153 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.3
  Found link https://files.pythonhosted.org/packages/31/f5/5d67634c2048874e68e64a789c6336fb00dea1d30b6226582323bfb9da4b/jetson-stats-4.2.4.tar.gz#sha256=63fa224b61c65d2faa31e5e9b3406e87164dffc472d3ec2c168480d9d0337020 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.4
  Found link https://files.pythonhosted.org/packages/eb/ff/9f259b170a0a09faea59db1ccabbeb95a9bd469960232c0277bfef68c7f5/jetson-stats-4.2.5.tar.gz#sha256=eb04d16048244177d60b30f7b4021a2e82adc8e60e53713eca7d6471eee3dbd2 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.5
  Found link https://files.pythonhosted.org/packages/73/7b/fef1e6ee102a75d780d8287b5db82b5cdc49f51ee76034a9f89a9ed18e05/jetson-stats-4.2.6rc1.tar.gz#sha256=350d2ac119edc967ad67e30a4ff2683071e72a5b721925ce49f7df6e2ea8c416 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.6rc1
  Found link https://files.pythonhosted.org/packages/c4/c8/6b70f60d25066361cc339e7bde50f32846fd7a80a56559678c9e51b8aa47/jetson-stats-4.2.6.tar.gz#sha256=43de8ee2ae29998fbcbd48324d78a68e16ecc2b3a2964ba8356f5b66f374ff03 (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.6
  Found link https://files.pythonhosted.org/packages/5f/78/1b8e2137f7d72014817ad18cef43b1ff0adc034d71a69de7d9b82603e939/jetson-stats-4.2.7.tar.gz#sha256=365990f8213e6a01748279b44827877c78b8796f52f3ab6d2832a3f5afc1390c (from https://pypi.org/simple/jetson-stats/) (requires-python:>=2.7), version: 4.2.7
Given no hashes to check 58 links for project 'jetson-stats': discarding no candidates
Installed version (4.2.7) is most up-to-date (past versions: 1.6.2, 1.6.3, 1.6.4, 1.6.5, 1.6.6, 1.6.7, 1.6.8, 1.6.9, 1.6.10, 1.6.11, 1.6.12, 1.6.13, 1.7.0, 1.7.1, 1.7.2, 1.7.3, 1.7.4, 1.7.5, 1.7.6, 1.7.7, 1.7.8, 1.7.9, 1.7.10, 1.7.11, 1.8.0, 1.8.1, 1.8.2, 1.8.3, 2.0.0, 2.0.1, 2.0.2, 2.0.3, 2.0.4, 2.1.0, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.1.4, 4.0.0, 4.1.0, 4.1.1, 4.1.2, 4.1.3, 4.1.4, 4.1.5, 4.2.0, 4.2.1, 4.2.2, 4.2.3, 4.2.4, 4.2.5, 4.2.6, 4.2.7)
Requirement already up-to-date: jetson-stats in /usr/local/lib/python3.8/dist-packages (4.2.7)
Requirement already satisfied, skipping upgrade: smbus2 in /usr/local/lib/python3.8/dist-packages (from jetson-stats) (0.4.3)
Requirement already satisfied, skipping upgrade: distro in /usr/lib/python3/dist-packages (from jetson-stats) (1.4.0)
Cleaning up...
Removed build tracker: '/tmp/pip-req-tracker-nr2xorpc'
```

### RAW Data

File from `jtop --error-log` attached

same output from `jetson_release -v`

rbonghi, 2024-05-07:
Hi @TakShimoda 

Can you try to reinstall the smbus python package with this command:

```
sudo pip3 install -U smbus2
```

BigNoseBob, 2024-05-21:
I have this as well!  I'm running a Jetson Nano 2GB dev kit w/ ubuntu 20.04

TakShimoda, 2024-05-30:
Hi @rbonghi sorry for the late response.

I tried your suggestion `sudo pip3 install -U smbus2` and get the following:

`Requirement already up-to-date: smbus2 in /usr/local/lib/python3.8/dist-packages (0.4.3)`

So nothing changes, even after rebooting. I don't recall what I did to get jtop initially. I believe it was already installed from the image that I burned from [here](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image)

daanbl, 2024-06-26:
I've encountered the same issue on a Jetson Xavier NX. Did anyone find a solution yet?

pawel-chmielak-deepsense, 2024-08-28:
I've had a similar issue on Jetson Nano (not Orin) and eventually got it working by forcibly reinstalling `smbus2` and `distro`, like that:
```
sudo pip3 install --force-reinstall -U smbus2
sudo pip3 install --force-reinstall -U distro
```

technobird22, 2024-09-22:
Thanks Pawel! Reinstalling worked :)

AmirYeganesahab, 2024-11-07:
On Jetson TX2NX https://github.com/rbonghi/jetson_stats/issues/512#issuecomment-2315175732 worked. Thanls Powel.

zszhere, 2024-12-25:
https://github.com/rbonghi/jetson_stats/issues/512#issuecomment-2315175732
On NVIDIA Orin NX this works well. Thanls Powel.

ByerRA, 2025-04-01:
I am having this same issue all of a sudden on both my Jetson Nano (8G) and Jetson Xavier NX (16G). [#512 (comment)](https://github.com/rbonghi/jetson_stats/issues/512#issuecomment-2315175732) didn't work for me.

derek-cook, 2025-06-09:
@ByerRA What worked for me is a force reinstall jetson-stats, after the previous commands mentioned:
```
sudo pip3 install --force-reinstall -U smbus2
sudo pip3 install --force-reinstall -U distro
sudo -H pip3 install --force-reinstall --no-cache-dir jetson-stats
```