rbonghi/jetson_stats #524: crash on start when run under console only without gui and mouse

project2go, 2024-05-20:
## Describe the bug

Jetson 6 on AGX orin, disable the GUI and running on SSH
seems mouse curses related issue. 

if you run jtop and keep pressing some keyboard  keys, it will start the jtop as normal, but if you don't press key fast enough, it will crash

## To Reproduce

Steps to reproduce the behavior:

1. sudo systemctl set-default multi-user.target
2. reboot
4. jtop

Error

jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.10/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 100, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 129, in run
    self.draw()
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 143, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/pall.py", line 147, in draw
    line_counter += self.process_table.draw(line_counter, 0, width, height_free_area, key, mouse)
AttributeError: 'ALL' object has no attribute 'process_table'

evildeeds, 2024-05-21:
This looks to be the same backtrace as #499 and #493

gztong, 2024-05-31:
This should be fixed by https://github.com/rbonghi/jetson_stats/commit/db70066fefe6b1fce92f7ce6a0a580797349dae0

manbehindthemadness, 2024-06-27:
I can confirm that this is fixed be switching to the develop branch past https://github.com/rbonghi/jetson_stats/commit/db70066fefe6b1fce92f7ce6a0a580797349dae0

rbonghi, 2024-07-25:
Fixed with the new release. Please update to **4.2.9**

```
sudo pip3 install -U jetson-stats
```

closed 2024-07-25 by rbonghi

gianpaj, 2025-05-31:
I'm having the same issue. I've an Nvidia Jetson Nano with Ubuntu 22.04

And `jtop` works only in the GUI interface. Not via SSH (openssh server)

```
sudo jtop
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.10/curses/__init__.py", line 73, in wrapper
    stdscr = initscr()
  File "/usr/lib/python3.10/curses/__init__.py", line 29, in initscr
    setupterm(term=_os.environ.get("TERM", "unknown"),
_curses.error: setupterm: could not find terminal
```

```
$ jtop --version
jtop 4.3.2
```

Is there a way to run `jtop` remotely? thanks