rbonghi/jetson_stats #527: Crash zero division error

deadmerc, 2024-06-04:
## Bug

```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.8/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.8/curses/__init__.py", line 105, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 100, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 129, in run
    self.draw()
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/jtopgui.py", line 143, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/pgpu.py", line 153, in draw
    chart.draw(self.stdscr, size_x, size_y, label=label_chart_gpu)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/lib/common.py", line 72, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/lib/chart.py", line 121, in draw
    self._plot_x_axis(stdscr, size_x, size_y, label=y_label)
  File "/usr/local/lib/python3.8/dist-packages/jtop/gui/lib/chart.py", line 170, in _plot_x_axis
    if ((point) / ceil(val)) % ten_sec == 0:
ZeroDivisionError: float modulo
```

## To Reproduce

Steps to reproduce the behavior:

1. run jtop
2. push button arrow right

## Expected behavior

Open GPU graph

### Board

Output from `jetson_release -v`:

* jetson-stats version: 4.2.6
* P-Number: p3767-0003
* Module: NVIDIA Jetson Orin Nano (8GB ram)
* Jetpack: 5.1.3
* L4T: 35.5.0

rbonghi, 2024-07-25:
I released the new jtop **4.2.9** let me know if fix this bug

```
sudo pip3 install -U jetson-stats
```

deadmerc, 2025-02-06:
Another one, when click on FAN directly after opening tool

```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.10/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 104, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 139, in run
    self.draw(page)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 155, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/pcontrol.py", line 412, in draw
    gui_chart['fan'][idx]['chart'].draw(self.stdscr, size_x, size_y, label=label_fan, y_label=False)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/lib/common.py", line 72, in wrapped
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/lib/chart.py", line 125, in draw
    self._plot_x_axis(stdscr, size_x, size_y, label=y_label)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/lib/chart.py", line 174, in _plot_x_axis
    if ((point) / ceil(val)) % ten_sec == 0:
ZeroDivisionError: float modulo
```

closed 2026-11-14 by johnnynunez