rbonghi/jetson_stats #534: Jtop Error only one module

VOCA-L, 2024-06-19:
## Describe the bug

I am using 5. Orin NX 8G model

It says that flashing is completed successfully in sdkmanager.

However, jtop fails like below only on one module. (screen shot)

I am using the same nvme by copying it, and I am curious why only one is showing this error.

note : 

## Screenshots

[image]

### Board
[image]

evildeeds, 2024-06-23:
Looks like the same backtrace as #492

rbonghi, 2024-07-25:
I think this may be related to this old PRs merged in `develop` and never published: #519 and #523
I'm still working on releasing a new jtop version that fixes all these bugs. otherwise let me know

rbonghi, 2024-07-25:
I released the new jtop **4.2.9** let me know if fix this bug

```
sudo pip3 install -U jetson-stats
```

closed 2024-09-24 by VOCA-L