rbonghi/jetson_stats #541: jetson-stats not supported for [L4T 32.7.5]

Coder1346, 2024-07-05:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 32.7.5

### Jetson-Stats

 - Version: 4.2.8

aksparab, 2024-07-11:
For me too its is showing 
[image]

Tzwjc, 2024-07-11:
[image]
[image]

Skylakc, 2024-07-18:
Same, the jtop 4.2.8 could not detect jetpack.

rbonghi, 2024-07-25:
Fixed with the new release. Please update to **4.2.9**

```
sudo pip3 install -U jetson-stats
```

closed 2024-07-25 by rbonghi

cuter9, 2024-07-28:
> Fixed with the new release. Please update to **4.2.9**
> 
> ```
> sudo pip3 install -U jetson-stats
> ```
 It works, thanks !