rbonghi/jetson_stats #546: ZeroDivisionError

Pecneb, 2024-07-11:
## Describe the bug

There is a potential zero division at `jtop/core/processes.py` line 110, in `get_process_info()`, that keeps crashing jtop.

## To Reproduce

Run jtop service -> sudo systemctl start jtop.service

## Screenshots

[image]

## Expected behavior

No crashes due to division by zero.

### Board

```
Software part of jetson-stats 4.2.8 - (c) 2024, Raffaello Bonghi
Model: Orin NX 16GB for DSBOARD-ORNXLAN - Jetpack 5.1.3 [L4T 35.5.0]
NV Power Mode[2]: 15W
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - Module: Check with sudo
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: P3768
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 Focal Fossa
 - Release: 5.10.192-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.8
 - Service: Inactive
Libraries:
 - CUDA: 11.4.315
 - cuDNN: 8.6.0.166
 - TensorRT: 8.5.2.2
 - VPI: 2.4.8
 - Vulkan: 1.3.204
 - OpenCV: 4.9.0 - with CUDA: NO
```

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

```
Jul 11 11:31:18 ax jtop[124507]: Process JtopServer-1:
Jul 11 11:31:18 ax jtop[124507]: Traceback (most recent call last):
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
Jul 11 11:31:18 ax jtop[124507]:     self.run()
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 466, in run
Jul 11 11:31:18 ax jtop[124507]:     if self._timer_reader.close(timeout=TIMEOUT_SWITCHOFF):
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 75, in close
Jul 11 11:31:18 ax jtop[124507]:     self._error_status()
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 90, in _error_status
Jul 11 11:31:18 ax jtop[124507]:     raise ex_value
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 46, in _timer_callback
Jul 11 11:31:18 ax jtop[124507]:     self._callback()
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 605, in jtop_stats
Jul 11 11:31:18 ax jtop[124507]:     data = self.jtop_decode()
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 569, in jtop_decode
Jul 11 11:31:18 ax jtop[124507]:     total, table = self.processes.get_status()
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 136, in get_status
Jul 11 11:31:18 ax jtop[124507]:     table = [self.get_process_info(prc[0], prc[3], prc[2], uptime) for prc in table]
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 136, in <listcomp>
Jul 11 11:31:18 ax jtop[124507]:     table = [self.get_process_info(prc[0], prc[3], prc[2], uptime) for prc in table]
Jul 11 11:31:18 ax jtop[124507]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 110, in get_process_info
Jul 11 11:31:18 ax jtop[124507]:     cpu_percent = 100 * (total_time / proc_uptime)
Jul 11 11:31:18 ax jtop[124507]: ZeroDivisionError: float division by zero
Jul 11 11:31:18 ax jtop[124488]: [INFO] jtop.service - Service closed
Jul 11 11:31:18 ax systemd[1]: jtop.service: Succeeded.
```

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -v -U jetson-stats`

### RAW Data

File from `jtop --error-log` attached

evildeeds, 2024-07-14:
This is the same issue as #522 which I provided a possible fix as well

rbonghi, 2024-07-25:
I think the bug found from @evildeeds also fix this issue

closed 2024-07-25 by rbonghi

rbonghi, 2024-07-25:
Fixed with the new release. Please update to **4.2.9**

```
sudo pip3 install -U jetson-stats
```