rbonghi/jetson_stats #555: The jtop service stoped suddenly[JP5.1.2]

iverson-fa, 2024-07-24:
## Describe the bug

The jtop service stoped suddenly.

## To Reproduce

Probabilistic occurrence
## Screenshots

If applicable, add screenshots to help explain your problem.

## Expected behavior

Jtop works properly.

### Board

Software part of jetson-stats 4.2.8 - (c) 2024, Raffaello Bonghi
Model: Jetson AGX Orin Developer Kit - Jetpack 5.1.2 [L4T 35.4.1]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - 699-level Part Number: 699-13701-0004-500 F.0
 - P-Number: p3701-0004
 - Module: NVIDIA Jetson AGX Orin (32GB ram)
 - SoC: tegra23x
 - CUDA Arch BIN: 8.7
 - Codename: Concord
Platform:
 - Machine: aarch64
 - System: Linux
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.120-tegra
 - Python: 3.8.10
jtop:
 - Version: 4.2.8
 - Service: Active
Libraries:
 - CUDA: 11.4.315
 - cuDNN: Not installed
 - TensorRT: Not installed
 - VPI: Not installed
 - Vulkan: 1.3.204
 - OpenCV: 4.2.0 - with CUDA: NO

### Log from jtop.service

Jul 24 18:54:24 home jtop[438247]: [CRITICAL] jtop.core.timer_reader - Exception in 'timer_reader thread': [Errno 3] No such process
Jul 24 18:54:26 home jtop[438247]: [ERROR] jtop.service - Error subprocess [Errno 3] No such process
Jul 24 18:54:26 home jtop[438247]: Traceback (most recent call last):
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 328, in run
Jul 24 18:54:26 home jtop[438247]:     control = self.q.get(timeout=timeout)
Jul 24 18:54:26 home jtop[438247]:   File "/usr/lib/python3.8/multiprocessing/queues.py", line 108, in get
Jul 24 18:54:26 home jtop[438247]:     raise Empty
Jul 24 18:54:26 home jtop[438247]: _queue.Empty
Jul 24 18:54:26 home jtop[438247]: During handling of the above exception, another exception occurred:
Jul 24 18:54:26 home jtop[438247]: Traceback (most recent call last):
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 451, in run
Jul 24 18:54:26 home jtop[438247]:     if self._timer_reader.close():
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 75, in close
Jul 24 18:54:26 home jtop[438247]:     self._error_status()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 90, in _error_status
Jul 24 18:54:26 home jtop[438247]:     raise ex_value
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 46, in _timer_callback
Jul 24 18:54:26 home jtop[438247]:     self._callback()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 605, in jtop_stats
Jul 24 18:54:26 home jtop[438247]:     data = self.jtop_decode()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 569, in jtop_decode
Jul 24 18:54:26 home jtop[438247]:       total, table = self.processes.get_status()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 136, in get_status
Jul 24 18:54:26 home jtop[438247]:     table = [self.get_process_info(prc[0], prc[3], prc[2], uptime) for prc in table]
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 136, in <listcomp>
Jul 24 18:54:26 home jtop[438247]:     table = [self.get_process_info(prc[0], prc[3], prc[2], uptime) for prc in table]
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/processes.py", line 94, in get_process_info
Jul 24 18:54:26 home jtop[438247]:     uid = int(cat(os.path.join('/proc', pid, 'loginuid')))
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/common.py", line 110, in cat
Jul 24 18:54:26 home jtop[438247]:     return f.readline().rstrip('\x00')
Jul 24 18:54:26 home jtop[438247]: ProcessLookupError: [Errno 3] No such process
Jul 24 18:54:26 home jtop[438247]: Process JtopServer-1:
Jul 24 18:54:26 home jtop[438247]: Traceback (most recent call last):
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 328, in run
Jul 24 18:54:26 home jtop[438247]:     control = self.q.get(timeout=timeout)
Jul 24 18:54:26 home jtop[438247]:   File "/usr/lib/python3.8/multiprocessing/queues.py", line 108, in get
Jul 24 18:54:26 home jtop[438247]:     raise Empty
Jul 24 18:54:26 home jtop[438247]: _queue.Empty
Jul 24 18:54:26 home jtop[438247]: During handling of the above exception, another exception occurred:
Jul 24 18:54:26 home jtop[438247]: Traceback (most recent call last):
Jul 24 18:54:26 home jtop[438247]:   File "/usr/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
Jul 24 18:54:26 home jtop[438247]:     self.run()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/service.py", line 466, in run
Jul 24 18:54:26 home jtop[438247]:     if self._timer_reader.close(timeout=TIMEOUT_SWITCHOFF):
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 75, in close
Jul 24 18:54:26 home jtop[438247]:     self._error_status()
Jul 24 18:54:26 home jtop[438247]:   File "/usr/local/lib/python3.8/dist-packages/jtop/core/timer_reader.py", line 90, in _error_status

rbonghi, 2024-07-25:
Fixed with the new release. Please update to **4.2.9**

```
sudo pip3 install -U jetson-stats
```

closed 2024-07-25 by rbonghi

razvanphp, 2025-06-26:
Is this problem related?

```
Jun 26 09:05:43 jtop[1682]: [INFO] jtop.service - Initialization service
Jun 26 09:05:43 jtop[1682]: Process JtopServer-1:
Jun 26 09:05:43 jtop[1682]: Traceback (most recent call last):
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/multiprocessing/process.py", line 313, in _bootstrap
Jun 26 09:05:43 jtop[1682]:     self.run()
Jun 26 09:05:43 jtop[1682]:     ~~~~~~~~^^
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/site-packages/jtop/service.py", line 210, in run
Jun 26 09:05:43 jtop[1682]:     data = self.jtop_decode()
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/site-packages/jtop/service.py", line 455, in jtop_decode
Jun 26 09:05:43 jtop[1682]:     total, table = self.processes.get_status()
Jun 26 09:05:43 jtop[1682]:                    ~~~~~~~~~~~~~~~~~~~~~~~~~^^
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/site-packages/jtop/core/processes.py", line 136, in get_status
Jun 26 09:05:43 jtop[1682]:     table = [self.get_process_info(prc[0], prc[3], prc[2], uptime) for prc in table]
Jun 26 09:05:43 jtop[1682]:              ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/site-packages/jtop/core/processes.py", line 94, in get_process_info
Jun 26 09:05:43 jtop[1682]:     uid = int(cat(os.path.join('/proc', pid, 'loginuid')))
Jun 26 09:05:43 jtop[1682]:               ~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jun 26 09:05:43 jtop[1682]:   File "/usr/lib/python3.13/site-packages/jtop/core/common.py", line 115, in cat
Jun 26 09:05:43 jtop[1682]:     with open(path, 'r') as f:
Jun 26 09:05:43 jtop[1682]:          ~~~~^^^^^^^^^^^
Jun 26 09:05:43 jtop[1682]: FileNotFoundError: [Errno 2] No such file or directory: '/proc/865/loginuid'
Jun 26 09:05:44 jtop[1665]: [INFO] jtop.service - Service closed
Jun 26 09:05:44 systemd[1]: jtop.service: Deactivated successfully.
```

jtop 4.3.1