rbonghi/jetson_stats #556: missing key freq and crash

rbonghi, 2024-07-31:
Key error on Freq CPU

Forum post: https://forums.developer.nvidia.com/t/orin-nx-jtop-read-cpu-crash/301611

[image]

soyszala, 2024-08-05:
Experienced the same issue, I'll post some details here in case you would find any information useful.

## Describe the bug

Jtop crashes when trying to check CPU information (switching to CPU tab)

## To Reproduce

1. Run `jtop` command
2. Press key 3

## Additional context

Always reproducible. Recently switched to new board and new Jetpack version (see below), didn't experience the issue before.

Traceback from the error:
```
Traceback (most recent call last):
  File "/usr/local/bin/jtop", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/jtop/__main__.py", line 160, in main
    curses.wrapper(JTOPGUI, jetson, pages, init_page=args.page,
  File "/usr/lib/python3.10/curses/__init__.py", line 94, in wrapper
    return func(stdscr, *args, **kwds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 100, in __init__
    self.run(loop, seconds)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 129, in run
    self.draw()
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/jtopgui.py", line 143, in draw
    page.draw(self.key, self.mouse)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/pcpu.py", line 138, in draw
    step_height, step_width, size_columns, size_rows = cpu_grid(
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/pcpu.py", line 78, in cpu_grid
    print_cpu(stdscr, idx, cpu, start_y + idx_row * step_height, start_x + idx_column * step_width, step_height - 1, step_width - 1)
  File "/usr/local/lib/python3.10/dist-packages/jtop/gui/pcpu.py", line 122, in print_cpu
    freq = cpu['freq']
KeyError: 'freq'
```

### Board

* jetson-stats version: 4.2.9
* P-Number: p3767-0000
* Module: NVIDIA Jetson Orin NX (16GB ram)
* Jetpack: 6.0
* L4T: 36.3.0

### Log from jtop.service

Error after `Aug 05 09:32:14` multiple times
<details>
<summary>LOG</summary>

```
Aug 05 09:32:08 tegra-ubuntu systemd[1]: Started jtop service.
Aug 05 09:32:08 tegra-ubuntu jtop[5921]: [INFO] jtop.service - jetson_stats 4.2.9 - server loaded
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.service - Running on Python: 3.10.12
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.hardware - Hardware detected aarch64
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.hardware - NVIDIA Jetson 699-level Part Number=699-13767-0000-300 M.1
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.hardware - NVIDIA Jetson Module=NVIDIA Jetson Orin NX (16GB ram)
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.hardware - NVIDIA Jetson detected L4T=36.3.0
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.cpu - Found 8 CPU
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.gpu - GPU "gpu" status in /sys/devices/platform/bus@0/17000000.gpu
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.gpu - GPU "gpu" frq in /sys/devices/platform/bus@0/17000000.gpu/devfreq/17000000.gpu
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.gpu - NVIDIA SMI exist!
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.processes - Process service started
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.memory - Found EMC!
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.memory - Memory service started
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.engine - Special Engine group found: [dlaX]
Aug 05 09:32:09 tegra-ubuntu jtop[5921]: [INFO] jtop.core.engine - Special Engine group found: [pvaX]
[... 274 lines cut ...]
  adding 'jtop/gui/lib/colors.py'
  adding 'jtop/gui/lib/common.py'
  adding 'jtop/gui/lib/linear_gauge.py'
  adding 'jtop/gui/lib/process_table.py'
  adding 'jtop/gui/lib/smallbutton.py'
  adding 'jetson_stats-4.2.9.dist-info/LICENSE'
  adding 'jetson_stats-4.2.9.dist-info/METADATA'
  adding 'jetson_stats-4.2.9.dist-info/WHEEL'
  adding 'jetson_stats-4.2.9.dist-info/entry_points.txt'
  adding 'jetson_stats-4.2.9.dist-info/top_level.txt'
  adding 'jetson_stats-4.2.9.dist-info/RECORD'
  removing build/bdist.linux-aarch64/wheel
  Building wheel for jetson-stats (setup.py) ... done
  Created wheel for jetson-stats: filename=jetson_stats-4.2.9-py3-none-any.whl size=155251 sha256=fccd6718ff96011b05c930943ef8611211805b90e00ea2c17a467226bb42013c
  Stored in directory: /tmp/pip-ephem-wheel-cache-ovk7od2y/wheels/b8/68/bf/2ba6de46fe5a7f601ae840db97f43e5681431d1028bde1c9fe
Successfully built jetson-stats
Installing collected packages: jetson-stats
  changing mode of /usr/local/bin/jetson_config to 755
  changing mode of /usr/local/bin/jetson_release to 755
  changing mode of /usr/local/bin/jetson_swap to 755
  changing mode of /usr/local/bin/jtop to 755
Successfully installed jetson-stats-4.2.9
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
```
</details>

soyszala, 2024-08-06:
One more thing:
I was experimenting with the power mode settings and the issue seems related to power modes.

By default my board had the `15W` power mode, which means, that (if I understand correctly), only 4 CPU cores were active out of 8. While this power mode was active, the crash described above always happened.
However, when I changed the power mode to `MAXN` the issue no longer occurred. In this power mode all the CPU cores were active in my case.

<details>
  <summary>Commands to change power modes</summary>
  
  Check current power mode
  ```
  sudo nvpmodel -q
  ```
  Set power mode
  ```
  sudo nvpmodel -m <x>
  ```
  Where `x` is the ID of the power mode, to list available power modes run:
  ```
  sudo nvpmodel -p --verbose
  ```
  
</details>

na-tt, 2024-08-19:
Experienced the same on an Nvidia Orin NX 16 GB developer kit with Jetpack 5.1.2 (L4T 35.4.1), following soyszala's advice and changing the power mode (which by default was 15W) to MAXN fixed the issue for me as well.

kentdotn, 2024-09-10:
I believe this problem occurs when some CPUs are offline.
I've fixed it with the following patch:
```diff
--- jtop/gui/pcpu.py     2024-09-10 05:59:46.172062761 +0000
+++ jtop/gui/pcpu.py.fixed       2024-09-10 05:59:27.839832496 +0000
@@ -119,13 +119,14 @@
         except curses.error:
             pass
         # Print info
-        freq = cpu['freq']
-        freq['online'] = cpu['online']
-        freq['name'] = "Frq"
-        try:
-            freq_gauge(stdscr, pos_y + size_h, pos_x, size_w, cpu['freq'])
-        except curses.error:
-            pass
+        if 'freq' in cpu:
+            freq = cpu['freq']
+            freq['online'] = cpu['online']
+            freq['name'] = "Frq"
+            try:
+                freq_gauge(stdscr, pos_y + size_h, pos_x, size_w, cpu['freq'])
+            except curses.error:
+                pass
 
     def draw(self, key, mouse):
         # Screen size
```
Now you can see no 'Frq` gauge for offline CPUs.
[image]

savf, 2024-09-26:
After the fix has been merged, the issue does no longer exist in jetson-stats 4.2.11. This issue can be closed.

rbonghi, 2024-09-26:
Thank you for your help! I really appreciate it!

kentdotn, 2024-09-26:
Great! Thank you!