rbonghi/jetson_stats #563: Installer hangs and service cannot be restarted

nsercproject, 2024-08-16:
## Describe the bug

The installer hangs while building on new Orin NX-based systems, using ConnetcTech's Photon carrier board (NGX020), L4T r35.4.1 BSP, Ubuntu 20.04, JP 5.1.2

## To Reproduce

Steps to reproduce the behaviour:

1. Run the following at the terminal: `sudo pip3 install -U jetson-stats`
2. The process hangs while the output to the terminal is: `Building wheel for jetson-stats (setup.py) ... -`
3. Try installing with a fresh download: `sudo pip3 install --no-cache-dir -v -U jetson-stats`
4. The process hangs while printing: ` INFO:jtop.service: - ENABLE jtop.service`
5. In both cases running `jtop` gives the following:
    `The jtop.service is not active. Please run:`
       `sudo systemctl restart jtop.service` 
6. Running the last line hangs. The service is enabled but inactive and can't be restarted.
7. Rebooting does not change the results of running all or some of the above commands

## Screenshots

## Expected behaviour

The installation should complete as it used to in previous systems using the same Photon board but the TX2 NX SoM. Actually this was working a month or so ago.

## Additional context

`jtop` is a nice tool for diagnosing and understanding the system responses when working with a new unit. 

### Board

Output from `jetson_release -v`:

* jetson-stats version: 4.2.8
* P-Number: p3767-0000
* Module: NVIDIA Jetson Orin NX (16GB ram)
* Jetpack: 5.1.2-b104
* L4T: R35 (release), REVISION: 4.1

### Log from jtop.service

Attach here the output from: `journalctl -u jtop.service -n 100 --no-pager`

$ journalctl -u jtop.service -n 100 --no-pager
-- Logs begin at Tue 2023-11-21 14:10:21 MST, end at Thu 2024-08-15 17:55:01 MDT. --
Aug 15 17:05:42 orin-4 systemd[1]: Failed to start jtop.service.
Aug 15 17:06:47 orin-4 systemd[1]: Failed to start jtop.service.
Aug 15 17:18:35 orin-4 systemd[1]: Failed to start jtop.service.
Aug 15 17:21:02 orin-4 systemd[1]: Failed to start jtop.service.

### Log from jetson-stats installation

Attach here the output from: `sudo -H pip3 install --no-cache-dir -v -U jetson-stats`

[sudo_H_install__no_cache_jetson_stats.txt](https://github.com/user-attachments/files/16631477/sudo_H_install__no_cache_jetson_stats.txt)

### RAW Data

File from `jtop --error-log` attached

vladmosoiu98, 2024-08-21:
I am currently having the same setup and encountered the exact same issue. 

Digging around the issues on this topic, I found a solution.
Basically when the install process is stuck at `INFO:jtop.service: - ENABLE jtop.service` in a separate terminal edit the `jtop.service` file that was just copied by the installer in `/etc/systemd/system/` and remove `After=multi-user.target`

Save it, then reload the systemd services with `systemctl daemon-reload` and restart the jtop.service

Now it should properly start and the installation to be finished.

I am still currently investigating why this happens, since the `After=multi-user.target` is a check has been added for almost a year and did not cause problems so far. Some users suggested it might be due to some issues related to the GUI coloring, but I am not sure.

The main source for this fix was this issue if it helps in any case: https://github.com/rbonghi/jetson_stats/issues/428#issuecomment-1627503419

Hopefully this might provide some insights for @rbonghi for a quickfix 😄

rbonghi, 2024-08-21:
Hi all!
Thank you for your posts! I will do it soon this week!

Quite exceptional busy this week at work and I didn't have time to work on jetson-stats

vladmosoiu98, 2024-08-21:
No worries, thanks for checking it.

After further inspection, it seems that `multi-user.target` depends to the `time-sync.target` which in my case had some configuration issue due to my `chrony` setup for a clock sync between multiple Orins.

So the quickfix I mentioned might not truly be universal, you should first check the `target` services status with `systemctl list-units --type=target` to make sure is related to this or not.

nsercproject, 2024-08-21:
[like]  Pablo Adames reacted to your message:

The main source for this fix was this issue if it helps in any case: #428 (comment)<https://github.com/rbonghi/jetson_stats/issues/428#issuecomment-1627503419>

9426224, 2025-05-15:
> I am currently having the same software setup (only using the Orin Nano DevKit) and encountered the exact same issue.
> 
> Digging around the issues on this topic, I found a solution. Basically when the install process is stuck at `INFO:jtop.service: - ENABLE jtop.service` in a separate terminal edit the `jtop.service` file that was just copied by the installer in `/etc/systemd/system/` and remove `After=multi-user.target`
> 
> Save it, then reload the systemd services with `systemctl daemon-reload` and restart the jtop.service
> 
> Now it should properly start and the installation to be finished.
> 
> I am still currently investigating why this happens, since the `After=multi-user.target` is a check has been added for almost a year and did not cause problems so far. Some users suggested it might be due to some issues related to the GUI coloring, but I am not sure.
> 
> The main source for this fix was this issue if it helps in any case: [#428 (comment)](https://github.com/rbonghi/jetson_stats/issues/428#issuecomment-1627503419)
> 
> Hopefully this might provide some insights for [@rbonghi](https://github.com/rbonghi) for a quickfix 😄

Thanks, after bunch of work, I delete `After=multi-user.target` and It really did work. Just don't know why, after version 4.2.0.dev1 all of this version cannot run on L4T 36.3.0 JetPack 6.0. Thanks for all!