rbonghi/jetson_stats #571: jetson-stats not supported for [L4T 35.6.0]

woodstockahem, 2024-09-19:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 35.6.0

### Jetson-Stats

 - Version: 4.2.10

SyWhs, 2024-09-20:
[image]
[image]
L4T: 35.6.0
Jetson-Stats：4.2.10
**

> Jetpack missing

**

skadiUnderTides, 2024-09-23:
same
[image]
[image]

rbonghi, 2024-09-23:
Fixed with the new release. Please update to **4.2.11**

```
sudo pip3 install -U jetson-stats
```

closed 2024-09-23 by rbonghi