rbonghi/jetson_stats #574: Power Monitoring for a Docker Container

nee1k, 2024-09-27:
Is there a way to monitor the power consumption of Docker containers running on Jetson devices using jtop.

I would like jtop to include container-specific power monitoring for CPU and GPU usage.

This feature would help users monitor power usage for containerized AI or edge applications on Jetson devices more efficiently.

rbonghi, 2024-09-27:
Hi @nee1k 

You can run jtop from Docker 

You can try this test:
1. Install jetson-stats on your host
2. Install jetson-stats on your container as well
3. Pass to your container /run/jtop.sock:/run/jtop.sock
```
docker run --rm -it -v /run/jtop.sock:/run/jtop.sock rbonghi/jetson_stats:latest
```

More details are here: https://rnext.it/jetson_stats/docker.html

nee1k, 2024-09-27:
Thank you for replying. 

Does the above method measure the power consumption of the entire jetson or just the docker container?

rbonghi, 2024-12-19:
The power consumption is from the whole Jetson. jtop reading the internal sensors of the board

vamsi2025nr, 2026-11-06:
The maintainer is right that these sensors read the whole module, so there is no per-container number to read. A workaround that stays honest: run one container at a time with a fixed workload, record module power over the run, and subtract an idle baseline measured the same way. Concurrent containers can't be separated like this. Also, tegrastats' --interval is only a print interval (default 1 s), jtop polls at a similar rate, and the INA3221's internal sampling rate is undocumented, so integrate over long runs rather than short ones.