rbonghi/jetson_stats #613: jetson-stats not supported for [L4T 35.6.1]

PRRH, 2025-03-19:
Please update jetson-stats with new jetpack

### Linux for Tegra

 - L4T: 35.6.1

### Jetson-Stats

 - Version: 4.3.1

deo-abhijit, 2025-03-20:
yep, recieved same warning. 

> [WARN] jetson-stats not supported for [L4T 35.6.1]
  Please, try: sudo pip3 install -U jetson-stats or
  open a Github issue (press CTRL + Click)

Jetson Version: jetson-stats-4.3.2

rbonghi, 2025-03-20:
Hi @deo-abhijit if you are working with jetson-stats 4.3.2 try:

```
sudo systemctl start jtop.service
```

Best,
Raffaello

chowned, 2025-04-04:
> Hi [@deo-abhijit](https://github.com/deo-abhijit) if you are working with jetson-stats 4.3.2 try:
> 
> ```
> sudo systemctl restart jtop.service
> ```
> 
> Best, Raffaello

**This solves the issue**, before this i run (xavier agx):

1. **dpkg -l | grep nvidia-jetpack** (i had no visibility of the installation, not sure why...)
2. **sudo apt install nvidia-jetpack**
3. **sudo pip3 install -U jetson-stats**
4. **sudo systemctl restart jtop.service**