rbonghi/jetson_stats #626: Can't run jtop on AGX orin with DRAM and eMMC changed(PCN210100) on Jetpack 6.2

redhao, 2025-05-30:
## Describe the bug
Can't run jtop on AGX orin with DRAM and eMMC changed(PCN210100) on Jetpack 6.2.
HINT:
The jtop.service in not active. Please run:
sudo systemctl  restart jtop.service.

kvnptl, 2025-07-01:
Yes, getting the same error

kvnptl, 2025-07-01:
Update: finally made it work!

The main issue was that the jtop service file was missing from `/etc/systemd/system`, copying it manually fixed the issue.

-> Follow this:

1. Uninstall and reinstall jetson-stats
```
sudo pip3 uninstall jetson-stats -y
sudo pip3 install jetson-stats
```

2. Manually install the jtop service (this was the key fix)
```bash
sudo cp /usr/local/jetson_stats/jtop.service /etc/systemd/system/
```

3. Reload systemd and enable/start the service
```bash
sudo systemctl daemon-reload
sudo systemctl enable jtop.service
sudo systemctl start jtop.service
````

4. Add user to jtop group for proper permissions
````
sudo usermod -a -G jtop $USER
````

5. Verify service is running
````
sudo systemctl status jtop.service
````
6. Test jtop
```
jtop
```

hemsinghb, 2025-07-14:
> What is the error?

the error is the "jtop is not compatible with jetpack 6.2.1."

redhao, 2025-07-14:
The error is following. Thanks.

BR,
Red